lvgl/lvgl #395: How to switch the theme while the program is running

qwert1213131, 2018-09-05:
For example, there are three themes of mono alien zen in the project. How can I switch freely?

kisvegabor, 2018-09-05:
There are two possible methods right now:
1. Delete all objects, call `lv_theme_set_current(new_theme)`, and create the objects again
2. Set the styles manually from the new theme.

However, I see that a function like `lv_theme_refresh(new_theme)` would be useful to update the theme in real time. I need to check what is the best way to implement it.

qwert1213131, 2018-09-05:
@kisvegabor 

I think that lv_theme_refresh would be more appropriate if you don't recreate the object. This is more suitable for theme switching.

kisvegabor, 2018-09-05:
I figured out a way to easily update the theme runtime. I implemented it in `dev-5.2`.
In `lv_conf.h` (in the theme section) you should add 
```c
#define LV_THEME_LIVE_UPDATE    1       /*Enable to change the theme in run time. Requires 8..10 kB extra RAM*/
```

And then try:
```c

    lv_theme_t * th1 = lv_theme_alien_init(210, NULL);
    lv_theme_t * th2 = lv_theme_zen_init(20, NULL);

    lv_test_theme_1(th1);
    lv_theme_set_current(th2);
```
`lv_examples` should be updated to `dev-5.2` too.

qwert1213131, 2018-09-06:
@kisvegabor 

I tested it and it works fine.
It’s great.
Thank you!

[image]

kisvegabor, 2018-09-06:
Awesome! Thank you for the test!
I see that you tested the new image converter tool as well!

closed 2018-09-06 by kisvegabor

colesnicov, 2019-03-20:
Hello, GREAT WORK! :)

It is not work for me.

- Device: esp32
- TFT: ILI9341 + Touch
- LVGL: 5.3.0
- Wrapper: [esp-iot](https://github.com/espressif/esp-iot-solution/tree/master/components/hmi/lvgl_gui)

lv_config.h:
`#define LV_THEME_LIVE_UPDATE    1`

and the code:

```
lv_theme_t *th;

switch (lv_ddlist_get_selected(ddlist)) {
    case 0:
	th = lv_theme_material_init(100, NULL);
	lv_theme_set_current(th);
	break;
}
```

But it does not do anything.

Of course, I do not use `lv_test_theme(...)`. I have got custom layout of UI..

kisvegabor, 2019-03-20:
Have look at this post: https://forum.littlevgl.com/viewtopic.php?p=250#p250

It seems there is a bug in ESP32's code and lv_conf is not included.