lvgl/lvgl #399: High .bss ram size

Masi2017, 2018-09-10:
Hi All,
I am using ESP32 and I ported lvgl. I just tested how much DRAM I am using for each components and noticed that liblvgl uses 35246 bytes from DRAM in .bss section. As per the lvgl document it is supposed to use less than 8kb. Would you please help me with that maybe I am setting something wrong.

kisvegabor, 2018-09-11:
Hi,

Have a look at this project: https://github.com/littlevgl/lv_projects/tree/master/esp32_ili9341

In general, you should lower `LV_MRM_SIZE` and `LV_VDB_SIZE` in `lv_conf.h`.

Masi2017, 2018-09-11:
Thanks kisvegbor . my lv_conf.h is same as this project but I want to use heap dynamic memory since I need to preserve DRAM for other purposes so I have to change LV_MEM_CUSTOM=1 to use dynamic memory. Also I wanna use external memory and because of that I replaced "heap_caps_malloc " function instead of malloc in LV_MEM_CUSTOM_ALLOC , in lv_mem.c , in lv_mem_alloc also I made the following changes:
/*use custom , user defined malloc function*/
/*Allocate a header too to store the size */
// alloc=LV_MEM_CUTOM_ALLOC ( size + sizeof (lv_mem_header_t));
alloc=LV_MEM_CUTOM_ALLOC ( size + sizeof (lv_mem_header_t),MALLOC_CAP_SPIRAM);

I am trying to replace malloc with heap_caps_malloc(size , MALLOC_CAP_SPIRAM) 
but system crashed , whould you please let me know what I am doing wrong?

Masi2017, 2018-09-11:
Thanks kisvegbor . My lv_conf.h was same as the project you have mentioned but I want to use dynamic memory since I need to preserve DRAM for other purposes so I changed LV_MEM_CUSTOM=1 to use dynamic memory. Also I wanna use external memory and because of that I replaced "heap_caps_malloc " function instead of malloc in LV_MEM_CUSTOM_ALLOC , in lv_mem.c , in lv_mem_alloc also I made the following changes:
/use custom , user defined malloc function/
/*Allocate a header too to store the size */
// alloc=LV_MEM_CUTOM_ALLOC ( size + sizeof (lv_mem_header_t));
alloc=LV_MEM_CUTOM_ALLOC ( size + sizeof (lv_mem_header_t),MALLOC_CAP_SPIRAM);

Now all memory has been placed in dynamic memory section. I was wondering if I can use dynamic memory instead of static memory for vdb?

kisvegabor, 2018-09-12:
> Now all memory has been placed in dynamic memory section. I was wondering if I can use dynamic memory instead of static memory for vdb?

It was already asked a few times in the near past. I will add it to `dev-5.2` soon.

kisvegabor, 2018-09-12:
I've just added it. In `lv_conf.h` set `LV_VDB_ADR` to `LV_VDB_ADR_INV`.

And before `lv_init()` do this:
```c
    uint32_t vs =  ((LV_VDB_SIZE * LV_VDB_PX_BPP) >> 3) + (((LV_VDB_SIZE * LV_VDB_PX_BPP) & 0x7) ? 1 : 0);
    printf("VDB size: %d\n", vs);
    void * vb = malloc(vs);            /*Replace with your custom malloc*/
    lv_vdb_set_adr(vb, NULL);
```
Please let me know how it works

andrebraga, 2018-09-12:
Oh just a hint: you'll regret allocating frame buffer memory on the SPI RAM as it's not DMA-able. In order to minimise CPU usage (which you'll want to) you definitely want to use DMA flushing. The IoT platform examples have ab almost functional implementation but you'll want to customise it a little further.

classic-gentleman, 2018-09-12:
Oh just a hint: you'll regret allocating frame buffer memory on the SPI RAM as it's not DMA-able. In order to minimise CPU usage (which you'll want to) you definitely want to use DMA flushing. The Espressif IoT platform examples have an almost functional implementation but you'll want to customise it just a little further.

classic-gentleman, 2018-09-12:
> I've just added it. In `lv_conf.h` set `LV_VDB_ADR` to `LV_VDB_ADR_INV`.

Maybe you committed locally but forgot to push to Github? I can't seem to find it on any branches... #nudge 😄

kisvegabor, 2018-09-12:
Upps, I really forget to push it. It should be fine now.

classic-gentleman, 2018-09-12:
Wonderful!

kisvegabor, 2018-09-12:
Does it mean you have tested it and it worked? :)

classic-gentleman, 2018-09-12:
Paraphrasing Knuth, _I only reviewed it correct, not tested it_ 😅

One thing I'm missing though to make it dummy-proof is exposing `((LV_VDB_SIZE * LV_VDB_PX_BPP) >> 3) + (((LV_VDB_SIZE * LV_VDB_PX_BPP) & 0x7) ? 1 : 0);` as a constant on `lv_conf.h` when `LV_VDB_ADR_INV` is used. Right now the only places where this is at all mentioned are this issue and `lv_vdb.c`. I noticed your alignment trick but I think it should be explicitly documented as such.

(Actually I'd much rather `#define LV_VDB_SIZE         (LV_VER_RES / LV_FB_SLICES * LV_HOR_RES)` and avoid blowing both my feet up by documenting the `LV_FB_SLICES` constant as an integer divisor of `LV_VER_RES`. What do you think?)

kisvegabor, 2018-09-14:
> One thing I'm missing though to make it dummy-proof is exposing ((LV_VDB_SIZE * LV_VDB_PX_BPP) >> 3) + (((LV_VDB_SIZE * LV_VDB_PX_BPP) & 0x7) ? 1 : 0); as a constant on lv_conf.h when LV_VDB_ADR_INV [...]

You are right it's quite ugly. I added `LV_VDB_SIZE_IN_BYTES` to `lv_vdb.h`.

> (Actually I'd much rather #define LV_VDB_SIZE (LV_VER_RES / LV_FB_SLICES * LV_HOR_RES) and avoid blowing both my feet up by documenting the LV_FB_SLICES constant as an integer divisor of LV_VER_RES. What do you think?)

Sorry, I don't understand. Maybe it's my poor Englis.   "avoid blowing both my feet up by documenting the LV_FB_SLICES". In other words? :)

Masi2017, 2018-09-14:
Hi All,
I had added some codes and could set a dynamic memory to work around my problem. I was wondering if I can clone version 5.2 or you are working on it?

kisvegabor, 2018-09-17:
@Masi2017 Yes,  you can clone it but I suggest to save your modified version before overwriting it.
Please let us know how it workd!

Masi2017, 2018-09-18:
@kisvegabor Ok, sure, I'll let you know.

kisvegabor, 2018-11-02:
As there was no activity here for a while I suppose the issue is solved.

If it's not working feel free the comment here and reopen this issue.

closed 2018-11-02 by kisvegabor