lvgl/lvgl #400: How to get hardware opacity?

vacore, 2018-09-11:
Hi!

For my hardware configuration each time I change the opacity style of the object (i.e. LV_OPA_50) the pixels that object is drawn with must contain it's alpha value in color.alpha member (not being mixed with background color, etc.), i.e in the 4th byte of the pixel, that gets sent to framebuffer.

What changes do I need to make to lvgl or its configuration to make that work?

kisvegabor, 2018-09-12:
Hi,

Do you mean something like this? 
https://github.com/littlevgl/lvgl/issues/262

vacore, 2018-09-12:
Not exactly. What I basically need is 3 opacity levels for different objects: full (for active screen), semi (for some background objects) and none (for active objects).

First I do as in the example you referenced.
lv_obj_set_style (lv_scr_act(), &lv_style_transp); and the screen indeed gets transparent.

Then I do
obj=lv_obj_create (lv_scr_act (), NULL);
lv_style_copy (&some_style, &lv_style_plain);
some_style.body.main_color = LV_COLOR_BLACK;
some_style.body.grad_color = LV_COLOR_BLACK;
some_style.body.opa = LV_OPA_70;
lv_label_set_style (hibar, &bar_style);

And no matter what opacity value I put into 'some_style.body.opa' it doesn't get respected and leaves the object completely opaque.

kisvegabor, 2018-09-12:
I see. The alpha channel of ARGB8888 can't be user directly right now but it should work:
1. In `lv_color.h` modify the `ret.alpha = 0xFF;` part in `lv_color_mix()` function to `ret.alpha = (uint16_t)((uint16_t) c1.alpha * mix + (c2.alpha * (255 - mix))) >> 8;`
2. Set `some_style.body.main_color.alpha = 0x30;`

vacore, 2018-09-12:
And that worked. Thank you very much!

kisvegabor, 2018-09-12:
Awesome! Your welcome!

closed 2018-09-12 by kisvegabor