lvgl/lvgl #401: Disable touch functionality for some objects

robrochta, 2018-09-11:
Is there a way to disable touch functionality for specific objects?
For example, I have a roller object with a button on the same screen.
I want to control selected position of the roller object with an encoder (and not to respond to a touch), and then select an action with a button. What would be the easiest way to do it?

kisvegabor, 2018-09-12:
Hi,
You can use `lv_obj_set_click(roller, false)`. However to the roller is compounded object you should disable the touch of the scrolling part too: `lv_obj_set_click(lv_page_get_scrl(roller), false)`

robrochta, 2018-09-12:
Thank you Gabor!
lv_obj_set_click(lv_page_get_scrl(roller), false) was the missing ingredient. Works exactly the way I wanted it to work.

closed 2018-09-12 by robrochta

seyyedemadmo, 2023-04-24:
this function dos not exist for me i use lvgl 8.2.0

kisvegabor, 2023-04-24:
In v8 you can use: `lv_obj_clear_flag(obj, LV_OBJ_FLAG_CLICKABLE);`