lvgl/lvgl #402: Suggestion - Widget - Animation using a range of bitmaps/images

ajithpv, 2018-09-12:
Hi,

I must say that, I'm new to the lvgl and going through the examples on PC.
I would like to know whether GIF is supported in lvgl? If not, then can we have some widget like below to do some basic animations with serious of pictures (like GIF).

- A widget where we specify the start and end images (assuming the image names are in sequential order).
- We can have a variable specifying the delay required between each image transfer (i.e. one image to next image).
- We can also have a variable holding the animation loop - whether we need to run the animation continuously or just 1 time.
-  Also, another variable for specifying the animation to be run in reverse or not (i.e. first the animation will start from 1st image to last image. Then it will continue from last image to 1st image).

Could you please let me now your thoughts on the same?

Regards,
Ajith P V

ajithpv, 2018-09-17:
Thank you @kisvegabor for your quick reply. I will check this out :+1:

embeddedt, 2018-11-10:
@kisvegabor For a simple animation like this I wonder if it is worth it to add a new object type. I feel that the actual code flow will be similar either way, because a new object will still have to change images in a similar manner. What do you think?

kisvegabor, 2018-11-10:
@embeddedt I agree, the example above should be suitable to create an image animation.