lvgl/lvgl #403: list: press and move down the list,  button action will be called.

fallstool, 2018-09-12:
Hello.
In the demo_create example,  the second part there a list.
when press and hold the first button (new), and then move the mouse to the next button (delete), the next button will pressed, when release the button, the button action (action3) will be called. is there any way that when move to the next button , the next button did not pressed.  
Thank You.

new   ->action 1
open  ->action 2
delete  ->action 3

kisvegabor, 2018-09-12:
Hi,

It should be fixed in `dev-5.2`.

Please update `lvgl` and `lv_examples` to `dev-5.2`.

fallstool, 2018-09-13:
Thank you! It works.

fallstool, 2018-09-13:
Hi, 
I test dev-5.2 version with the list.
There is an other problem. If the list go top, when I press the button(new) and move down, when the finger out of the range of the button(new), the button(new) state is also pressed, and release will call the button's action(action 1).  I suggest if the finger's postion is out of the button's range, I will reverse the button state to release, and will not call the button action.
Thank you!

kisvegabor, 2018-09-13:
You are right. I fixed it in `dev-5.2`. Test it, please!

fallstool, 2018-09-14:
Hi,
I tested it working fine. thank you!

closed 2018-09-14 by fallstool