lvgl/lvgl #404: about input device

792499568, 2018-09-12:
I have a question about the input device, and I use two key inputs. 
I want to achieve page turning function, two keys corresponding to the previous page and the next page. 
However, I found that the function: 
`lv_indev_t * lv_indev_drv_register(lv_indev_drv_t *driver);`
and the struct: 
`struct _lv_obj_t;
struct _lv_group_t;

/*The main input device descriptor with driver, runtime data ('proc') and some additional information*/
typedef struct _lv_indev_t {
    lv_indev_drv_t driver;
    lv_indev_proc_t proc;
    uint32_t last_activity_time;
    union {
        struct _lv_obj_t *cursor;       /*Cursor for LV_INPUT_TYPE_POINTER*/
        struct _lv_group_t *group;      /*Keypad destination group*/
        lv_point_t * btn_points;      /*Array points assigned to the button ()screen will be pressed here by the buttons*/

    };
    struct _lv_indev_t *next;
} lv_indev_t;`
does not provide the function of calling other functions after pressing the button, like the button:
`void lv_btn_set_action(lv_obj_t * btn, lv_btn_action_t type, lv_action_t action);`

Are there any relevant examples for reference?

kisvegabor, 2018-09-13:
Hi,

You can do this in your `indev_read` function. So if your "left" or "right" button is pressed you can call a LittelvGL function to do something.

792499568, 2018-09-17:
Thanks for your reply,

I solved this problem by binding buttons (physical) to the built-in button group.

> lv_obj_t * btn_next = lv_btn_create(lv_scr_act(), NULL);
    **lv_btn_set_action(btn_next, LV_BTN_ACTION_CLICK, kb_next);**
    lv_btn_set_fit(btn_next, true, true);
    lv_obj_t * l = lv_label_create(btn_next, NULL);
    lv_label_set_text(l, ">");
    lv_obj_align(btn_next, NULL, LV_ALIGN_IN_BOTTOM_LEFT, LV_DPI / 4, - LV_DPI / 4);
    **lv_group_add_obj(g1, btn_next);**

> uint8_t key_press(void)
{
    if(KEY1==1 || KEY2==1)
    {
        delay_ms(20);
        if(KEY1==1)
        {
            state = LV_INDEV_STATE_PR;
            **lv_group_send_data(g1, LV_GROUP_KEY_ENTER);**
            return LV_GROUP_KEY_ENTER;
        }else if(KEY2==1)
        {
            state = LV_INDEV_STATE_PR;
            **lv_group_send_data(g2, LV_GROUP_KEY_ENTER);**
            return LV_GROUP_KEY_ENTER;
        }
        state = LV_INDEV_STATE_REL;
        return 0;
    }
}

Then I had an unsolvable problem, The button will have an orange border to indicate focus, which I can't modify or remove.
I tried to change the “style” of the group, but it didn't work, Is there any way to modify or remove this border?

kisvegabor, 2018-09-17:
Great! It's also a possible solution.

To remove the orange recolor set a custom `style_mod` function which does nothing: 

```c
lv_group_set_style_mod_cb(g1, style_mod_nothing);

...

void style_mod_nothing(lv_style_t * style)
{
    /*Do nothing. Not change the style*/
}

```

kisvegabor, 2018-10-07:
As there was no activity here for a while  I suppose the issue is solved.

If it's not working feel free the comment here and reopen this issue.

closed 2018-10-07 by kisvegabor