lvgl/lvgl #407: Excessive calls to img_decoder (PNG, lodepng)

aenchevich, 2018-09-13:
Hi Gabor

I am so happy that there is now PNG support! I do see something that looks like a bug and I am reporting it here, but overall - yay, great job!

--------------------------------------

So...

I followed the steps in https://littlevgl.com/blog/50/use-png-images-in-littlevgl and got the same visual results (the 4 dice are displayed over the blueish background, with transparency).

However I a get this output (I put printf's to the decoder functions to see when they are called. This is ran on the PC btw):

png_decoder_info called
png_decoder_info called
png_decoder_open called
png_decoder_close called
png_decoder_info called
png_decoder_open called
png_decoder_close called
png_decoder_info called
png_decoder_open called
png_decoder_close called
png_decoder_info called
png_decoder_open called
png_decoder_close called

As you can see, firstly png_decoder_info is called once and then the entire sequence (info-open-close) is repeated 4 times. This does not seem correct?

I am attaching my code, although it is pretty much the same as the tutorial...

[decoder.cc.txt](https://github.com/littlevgl/lvgl/files/2379409/decoder.cc.txt)
[main.c.txt](https://github.com/littlevgl/lvgl/files/2379410/main.c.txt)

kisvegabor, 2018-09-13:
Hi,

Thank you for testing! It's so good to hear that it work for somebody else too :)

What is your `LV_VDB_SIZE`?
I suspect that it's because LittelvGL can draw the image in 4 parts due to `LV_VDB_SIZE`. 
If you set `LV_VDB_SIZE   (LV_HOR_RES * LV_VER_RES)` open and close should be called only once.

aenchevich, 2018-09-19:
Initially it was:

`#define LV_VDB_SIZE         ((LV_VER_RES * LV_HOR_RES)/9)`

I tried your suggestion but it did not improve the situation.

On top of that, the image does not show correctly on my Broadcom device:
 in the upper left corner there is a white rectangle showing a small part of the "No data.." message, usually displayed when the image cannot be found. I am attaching a screenshot. The rest of the image on the screen is a video stream.

[image]

kisvegabor, 2018-09-19:
Hi,

Please turn on the logging in lv_conf.h by: 
```c
#define LV_LOG_LEVEL	LV_LOG_LEVEL_INFO
#define LV_LOG_PRINTF	1
```
It should show which part causes  "No data".

The transparency looks ugly in your case because the not fully transparent pixels (e.g. the pixels in the holes of the dices) are mixed with the background. And in your case, the background is black but transparent. However, it's true that the drawing should take into account the alpha channel of ARGB888 pixels.

aenchevich, 2018-09-20:
Hi Gabor

About the "No data" chunk:
--------------------------------------
This is the stack trace of the call to lv_draw_img which causes 'No data":

```
(gdb) bt
#0  lv_draw_img (coords=0x49afb8 <work_mem_int+440>, mask=0x6eceece4, src=0x0, style=0x4a4450 <lv_style_plain>, opa_scale=255 '\377') at lvgl/lv_draw/lv_draw_img.c:69
#1  0x0042f344 in lv_img_design (img=0x49afa8 <work_mem_int+424>, mask=0x6eceece4, mode=0 '\000') at lvgl/lv_objx/lv_img.c:282
#2  0x0042b7e0 in lv_refr_obj (obj=0x49afa8 <work_mem_int+424>, mask_ori_p=0x6eceed4c) at lvgl/lv_core/lv_refr.c:507
#3  0x0042b96c in lv_refr_obj (obj=0x49ae90 <work_mem_int+144>, mask_ori_p=0x6eceedb8) at lvgl/lv_core/lv_refr.c:532
#4  0x0042b504 in lv_refr_obj_and_children (top_p=0x49ae90 <work_mem_int+144>, mask_p=0x6eceedb8) at lvgl/lv_core/lv_refr.c:442
#5  0x0042b23c in lv_refr_area_part_vdb (area_p=0x462820 <inv_buf>) at lvgl/lv_core/lv_refr.c:381
#6  0x0042b02c in lv_refr_area_with_vdb (area_p=0x462820 <inv_buf>) at lvgl/lv_core/lv_refr.c:337
#7  0x0042acac in lv_refr_areas () at lvgl/lv_core/lv_refr.c:251
#8  0x0042a844 in lv_refr_task (param=0x0) at lvgl/lv_core/lv_refr.c:177
#9  0x00430f5c in lv_task_exec (lv_task_p=0x49ae70 <work_mem_int+112>) at lvgl/lv_misc/lv_task.c:301
#10 0x004306d0 in lv_task_handler () at lvgl/lv_misc/lv_task.c:117
#11 0x00421598 in gui_main_thread (warning: GDB can't find the start of the function at 0x77b9f1e7.

```
You can assume that my lvgl is at BRANCH: dev-5.2; COMMIT: "enum refactor fixes" (839757e6c88f23d79f0cb9a2bc0d93a568683036)
In reality it is one commit above it, at my pull-request "Adding missing #ifdef LV_CONF_INCLUDE_SIMPLE", but this commit only adds the ifdef to v_theme_nemo.h.

About the "ugly" black transparency
----------------------------------------------
Is this due to a combination of two factors: a) at present transparency is implemented via the hack in the ..._flush function + b) there are "semi-transparent" pixels in the image? So the fully transparent pixels (full-black) are ok but "almost transparent" pixels show up as the ugly black border?

Would you like me to open a new issue about the transparency problem?

aenchevich, 2018-09-20:
I think the "No data" problem is a timing issue of some sorts and that it is related to the excessive calls to the png decoder. 

Here's why:
a) When I run the code on the PC I also see the "No data" problem once every 10-15 runs. Obviously yesterday the problem did not manifest itself the first time I ran the code on the PC, so I said "Ok, great, it works" and I proceeded on to test on my real hardware, where the problem happens always and that's why I saw it there but not on the PC. 

b) As I showed in my initial message, the "png_decoder_info - png_decoder_open - png_decoder_close" sequence is called 4 times. However sometimes, it is called 3 times. So..
 - When it is called 4 times the problem DOES NOT happen.
 - When it is called 3 times the problem HAPPENS.

**It appears to me as if the last, 4th call to the decoder (when there is a 4th call) draws everything correctly, on top of a previously incorrectly drawn "No data".** 

Here's a screenshot of the problem on PC :
[image]

kisvegabor, 2018-09-21:
**Ugly transparency**
I pushed an update the calculate the transparency correctly. (dev-5.2) In lv_conf.h you need to: `#define LV_COLOR_SCREEN_TRANSP    1`. It also includes the current "hack" in disp_flush. (Now it is in `lv_flush_ready()`) After enabling this feature you can set: `lv_style_screen.body.opa = LV_OPA_TRANSP` or to any opa value like `LV_OPA_50`.

**No data**
I tried your decoder functions and set LV_VDB_SIZE to (LV_HOR_RES * LV_VER_RES).  Interestingly I'm experiencing neither the "No data" nor the too much call of decoder functions issue. My console log is just: 
```
png_decoder_info called
png_decoder_info called
png_decoder_open called
png_decoder_close called
```

From the line in your trace, it seems the image source is NULL in the first call but after it's ok. It's also strange. Please compile with 
```c
#define LV_LOG_LEVEL	LV_LOG_LEVEL_INFO
#define LV_LOG_PRINTF	1
```
to see the internal errors of LittelvGL.

aenchevich, 2018-09-21:
Hi Gabor

Thank you for the update on transparency!

Regarding the "No data" problem: Yes, I did compile with these 2 defines (the first time you asked), but there was nothing interesting in the output, except the location of the "No data" so I did not say anything. I set the break point at the location of the "No data" message, provided by the log and got the stack trace above. There were no LittlevGL errors.

kisvegabor, 2018-09-23:
Hi Alex,

Transparency: Your welcome!  Have you had a chance to test it as well? 

No data: From the stack trace I can't see exactly what's going on because something went wrong earlier. So the "No data" is just a result of a previous failure. I added some new logs to `lv_img` hopefully they will help to find the root of the issue. BTW, I saw that you slightly modified the decoder functions from the blog post. Do you get the same result with the original functions too?

kisvegabor, 2018-10-07:
As there was no activity here for a while I close this issue.

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2018-10-07 by kisvegabor