lvgl/lvgl #411: Glyph margins from font_conv_core.php

vacore, 2018-09-16:
Hi!

Outputting time with font C file, obtained with font_conv_core.php, to make sure colon (:) looked good, I had no better idea, than manually tailoring the .c file adding and deleting space from sides of the glyph, thus regulating where it is placed.

Which in turn, kicked the fonts out of the build system that generated it from files with parameters for fonts.

How do you suggest to fix that?

May I propose to add enhancements for font_conv_core.php's interface, that different parameters were specified for each glyph in the font, such as margins as in the case above, and other glyphs attributes?

By the way, how do I add various Unicode symbols to the list parameter, like U+0E82 ?

kisvegabor, 2018-09-19:
Hi,

1. You can adjust the spacing in the style with `style.text.letter_space`. However, it sets the letter space in general for every character in the text.

2. You can create 3 labels: hours + ":" + minutes and position them individually.

3. You can manually set the width of the "space" character to 1. It enables you to set add the required spacing in the text.

Do the above mentioned suggestion applicable in your case? 
I can hardly imagine a user-friendly interface which allows setting margins manually for each character.

To add U+0E82 you type: `&#x0E82;` as in HTML.

vacore, 2018-09-19:
Thank you for your answer!

I've tried pp. 1&2 and got more or less decent results. In the end we decided not to blink the colon whatsoever as no modern UI does this and now it looks good.

Nevertheless I feel like it'd be great if I could control font generation in more detail. I suggest you have a look at emWin's FontConverter, how it's done there if you haven't done already. Font parameters for each glyph could be stored in a file, which is fed to the php script to generate a font. Those files could be generated from a script and tailored manually, or generated by a UI program, similar to the FontConverter's. Thanks.