lvgl/lvgl #418: Is there a way to rotate a image?

mianhuacar, 2018-09-20:
Hi,
Is there a way to rotate a image? I want to make a knob like below :)
[image]

And i want to make clock too :)
[image]

Thank you

kisvegabor, 2018-09-21:
Hi,

Unfortunately, image rotation is not supported now. 

It would be required for the clock indeed. However, why is it required for the knob? You can have the whole knob as an image except the little "circle". The little "circle" can be an other image which just needs to be positioned on the background knob.

mianhuacar, 2018-09-21:
Oh, thank you.
I will waiting the day lvgl can rotate the image :)

kisvegabor, 2018-09-23:
Me too. :)

I have an intention to add an image rotate feature once but unfortunately, I can't promise anything about right now.

Anyway, I leave open this issue as a reminder.

mianhuacar, 2018-09-24:
Thanks a lot

kisvegabor, 2018-11-30:
As adding image rotate is not planned in the near feature, to keep the list of issues I close it.

Image rotate and image zoom is the two missing advanced features so I'm sure they won't be forgotten. :)

closed 2018-11-30 by kisvegabor