lvgl/lvgl #419: Button Matrix map id?

rockcastle, 2018-09-20:
Hi,
My question is I could not find button matrix map id. Is there any way to do it?
btnm_str = {"a", "b", "c", "d", ""}
is there any way to go to like btnm_map_id[2] to go to "c"?

Thanks

kisvegabor, 2018-09-21:
Hi,

How do you mean "could not find button matrix map id"? 
Where would you like to find it?

rockcastle, 2018-09-21:
It is a suggestion if it is possible. There is no btnm_map_id now.

kisvegabor, 2018-09-23:
Do you mean in the action function? 
Please specify your use-case or give a simple example code where we can see how you would like to use the `map_id.

flashdark, 2018-09-29:
If you are looking to get the character string(s) i think this works.
  const  char  ** lv_btnm_get_map(lv_obj_t * btnm);  // Get the current map of a button matrix

If you are looking to toggle or press a button on the map to call the action function, you might try using the above to select the char string and the just call the action function directly.

kisvegabor, 2018-11-02:
As there was no activity here for a while I close this issue.

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2018-11-02 by kisvegabor