lvgl/lvgl #423: get someting wrong with example lv_apps/demo dev5.2

mianhuacar, 2018-09-23:
Hi,
I get someting wrong with example lv_apps/demo dev5.2. When press the ta, the keyboard goes out, but it will go into a dead while. No promble in 5.1.
[image]

Thank you:)

kisvegabor, 2018-09-23:
Hi, 

I tested and it woks for me on PC. 
Please enable logging in `lv_conf.h` and post the result:
```c
/*Log settings*/
#define USE_LV_LOG		1	/*Enable/disable the log module*/
#if USE_LV_LOG
/* How important log should be added:
 * LV_LOG_LEVEL_TRACE		A lot of logs to give detailed information
 * LV_LOG_LEVEL_INFO		Log important events
 * LV_LOG_LEVEL_WARN		Log if something unwanted happened but didn't caused problem
 * LV_LOG_LEVEL_ERROR		Only critical issue, when the system may fail
 */
#define LV_LOG_LEVEL	LV_LOG_LEVEL_INFO
/* 1: Print the log with 'printf'; 0: user need to register a callback*/

#define LV_LOG_PRINTF	1
```

If you can debug then a stack trace or at least the line where the code stops would be very useful.

mianhuacar, 2018-09-24:
hi kisvegabor,
#define LV_LOG_LEVEL	LV_LOG_LEVEL_INFO
----------------------------------------------------------
Info: Screen create ready 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #180)
Info: Screen create ready 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #180)
Info: Screen create ready 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #180)
Info: lv_init ready 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #108)
Info: button matrix created 	(external/lvgl-5.2/lvgl/lv_objx/lv_btnm.c #127)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: tab view created 	(external/lvgl-5.2/lvgl/lv_objx/lv_tabview.c #177)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: page created 	(external/lvgl-5.2/lvgl/lv_objx/lv_page.c #145)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: page created 	(external/lvgl-5.2/lvgl/lv_objx/lv_page.c #145)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: page created 	(external/lvgl-5.2/lvgl/lv_objx/lv_page.c #145)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: page created 	(external/lvgl-5.2/lvgl/lv_objx/lv_page.c #145)
Info: label created 	(external/lvgl-5.2/lvgl/lv_objx/lv_label.c #129)
Info: text area created 	(external/lvgl-5.2/lvgl/lv_objx/lv_ta.c #171)
Info: button matrix created 	(external/lvgl-5.2/lvgl/lv_objx/lv_btnm.c #127)
Info: keyboard created 	(external/lvgl-5.2/lvgl/lv_objx/lv_kb.c #138)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: page created 	(external/lvgl-5.2/lvgl/lv_objx/lv_page.c #145)
Info: list created 	(external/lvgl-5.2/lvgl/lv_objx/lv_list.c #143)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: button created 	(external/lvgl-5.2/lvgl/lv_objx/lv_btn.c #152)
Info: image created 	(external/lvgl-5.2/lvgl/lv_objx/lv_img.c #106)
Info: label created 	(external/lvgl-5.2/lvgl/lv_objx/lv_label.c #129)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: button created 	(external/lvgl-5.2/lvgl/lv_objx/lv_btn.c #152)
Info: image created 	(external/lvgl-5.2/lvgl/lv_objx/lv_img.c #106)
Info: label created 	(external/lvgl-5.2/lvgl/lv_objx/lv_label.c #129)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: button created 	(external/lvgl-5.2/lvgl/lv_objx/lv_btn.c #152)
Info: image created 	(external/lvgl-5.2/lvgl/lv_objx/lv_img.c #106)
Info: label created 	(external/lvgl-5.2/lvgl/lv_objx/lv_label.c #129)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: button created 	(external/lvgl-5.2/lvgl/lv_objx/lv_btn.c #152)
Info: image created 	(external/lvgl-5.2/lvgl/lv_objx/lv_img.c #106)
Info: label created 	(external/lvgl-5.2/lvgl/lv_objx/lv_label.c #129)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: button created 	(external/lvgl-5.2/lvgl/lv_objx/lv_btn.c #152)
Info: image created 	(external/lvgl-5.2/lvgl/lv_objx/lv_img.c #106)
Info: label created 	(external/lvgl-5.2/lvgl/lv_objx/lv_label.c #129)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: button created 	(external/lvgl-5.2/lvgl/lv_objx/lv_btn.c #152)
Info: image created 	(external/lvgl-5.2/lvgl/lv_objx/lv_img.c #106)
Info: label created 	(external/lvgl-5.2/lvgl/lv_objx/lv_label.c #129)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: button created 	(external/lvgl-5.2/lvgl/lv_objx/lv_btn.c #152)
Info: image created 	(external/lvgl-5.2/lvgl/lv_objx/lv_img.c #106)
Info: label created 	(external/lvgl-5.2/lvgl/lv_objx/lv_label.c #129)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: label created 	(external/lvgl-5.2/lvgl/lv_objx/lv_label.c #129)
Info: mesasge box created 	(external/lvgl-5.2/lvgl/lv_objx/lv_mbox.c #119)
Info: button matrix created 	(external/lvgl-5.2/lvgl/lv_objx/lv_btnm.c #127)
Info: chart created 	(external/lvgl-5.2/lvgl/lv_objx/lv_chart.c #118)
Info: bar created 	(external/lvgl-5.2/lvgl/lv_objx/lv_bar.c #103)
Info: slider created 	(external/lvgl-5.2/lvgl/lv_objx/lv_slider.c #107)

mianhuacar, 2018-09-24:
#define LV_LOG_LEVEL    LV_LOG_LEVEL_TRACE
------------------------------------------------------
Trace: lv_init started 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #65)
Trace: Screen create started 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #128)
Info: Screen create ready 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #180)
Trace: Screen create started 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #128)
Info: Screen create ready 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #180)
Trace: Screen create started 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #128)
Info: Screen create ready 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #180)
Info: lv_init ready 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #108)
Trace: tab view create started 	(external/lvgl-5.2/lvgl/lv_objx/lv_tabview.c #69)
Trace: Object create started 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #184)
Trace: button matrix create started 	(external/lvgl-5.2/lvgl/lv_objx/lv_btnm.c #67)
Trace: Object create started 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #184)
Info: button matrix created 	(external/lvgl-5.2/lvgl/lv_objx/lv_btnm.c #127)
Trace: Object create started 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #184)
Trace: container create started 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #67)
Trace: Object create started 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #184)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: tab view created 	(external/lvgl-5.2/lvgl/lv_objx/lv_tabview.c #177)
Trace: page create started 	(external/lvgl-5.2/lvgl/lv_objx/lv_page.c #60)
Trace: container create started 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #67)
Trace: Object create started 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #184)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Trace: container create started 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #67)
Trace: Object create started 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #184)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: page created 	(external/lvgl-5.2/lvgl/lv_objx/lv_page.c #145)
Trace: page create started 	(external/lvgl-5.2/lvgl/lv_objx/lv_page.c #60)
Trace: container create started 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #67)
Trace: Object create started 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #184)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Trace: container create started 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #67)
Trace: Object create started 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #184)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: page created 	(external/lvgl-5.2/lvgl/lv_objx/lv_page.c #145)
Trace: page create started 	(external/lvgl-5.2/lvgl/lv_objx/lv_page.c #60)
Trace: container create started 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #67)
Trace: Object create started 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #184)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Trace: container create started 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #67)
Trace: Object create started 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #184)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: page created 	(external/lvgl-5.2/lvgl/lv_objx/lv_page.c #145)
Trace: text area create started 	(external/lvgl-5.2/lvgl/lv_objx/lv_ta.c #78)
Trace: page create started 	(external/lvgl-5.2/lvgl/lv_objx/lv_page.c #60)
Trace: container create started 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #67)
Trace: Object create started 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #184)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Trace: container create started 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #67)
Trace: Object create started 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #184)
Info: container created 	(external/lvgl-5.2/lvgl/lv_objx/lv_cont.c #108)
Info: page created 	(external/lvgl-5.2/lvgl/lv_objx/lv_page.c #145)
Trace: label create started 	(external/lvgl-5.2/lvgl/lv_objx/lv_label.c #67)
Trace: Object create started 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #184)
Info: label created 	(external/lvgl-5.2/lvgl/lv_objx/lv_label.c #129)
Trace: animation create started 	(external/lvgl-5.2/lvgl/lv_misc/lv_anim.c #65)
Trace: animation created 	(external/lvgl-5.2/lvgl/lv_misc/lv_anim.c #85)
Info: text area created 	(external/lvgl-5.2/lvgl/lv_objx/lv_ta.c #171)
Trace: animation create started 	(external/lvgl-5.2/lvgl/lv_misc/lv_anim.c #65)
Trace: animation created 	(external/lvgl-5.2/lvgl/lv_misc/lv_anim.c #85)
Trace: keyboard create started 	(external/lvgl-5.2/lvgl/lv_objx/lv_kb.c #78)
Trace: button matrix create started 	(external/lvgl-5.2/lvgl/lv_objx/lv_btnm.c #67)
Trace: Object create started 	(external/lvgl-5.2/lvgl/lv_core/lv_obj.c #184)
Info: button matrix created 	(external/lvgl-5.2/lvgl/lv_objx/lv_btnm.c #127)
Info: keyboard created 	(external/lvgl-5.2/lvgl/lv_objx/lv_kb.c #138)
Trace: animation create started 	(external/lvgl-5.2/lvgl/lv_misc/lv_anim.c #65)
Trace: animation created 	(external/lvgl-5.2/lvgl/lv_misc/lv_anim.c #85)
Trace: list create started 	(external/lvgl-5.2/lvgl/lv_objx/lv_list.c #72)
Trace: page create started 	(external/lvgl-5.2/lvgl/lv_objx/lv_page.c #60)
[... 458 lines cut ...]
Trace: animation create started 	(external/lvgl-5.2/lvgl/lv_misc/lv_anim.c #65)
Trace: animation created 	(external/lvgl-5.2/lvgl/lv_misc/lv_anim.c #85)
[... cut ...]
Trace: idnev read started 	(external/lvgl-5.2/lvgl/lv_hal/lv_hal_indev.c #115)
Trace: idnev read finished 	(external/lvgl-5.2/lvgl/lv_hal/lv_hal_indev.c #117)
Trace: indev task finished 	(external/lvgl-5.2/lvgl/lv_core/lv_indev.c #309)
Trace: display refresh task started 	(external/lvgl-5.2/lvgl/lv_core/lv_refr.c #183)
Trace: disp flush  started 	(external/lvgl-5.2/lvgl/lv_hal/lv_hal_disp.c #157)
Trace: disp flush ready 	(external/lvgl-5.2/lvgl/lv_hal/lv_hal_disp.c #159)
Trace: display refresh task finished 	(external/lvgl-5.2/lvgl/lv_core/lv_refr.c #204)

kisvegabor, 2018-09-24:
From the log all seems fine. Can you see in debugger where does the program halt?

mianhuacar, 2018-09-24:
Yes, very strange. I have no tool to see in debugger where does the program halt. And i found it may halt when loading a big image bin(about 1.5Mb) today. There is no promblem in 5.1.

kisvegabor, 2018-09-24:
How do you mean "bin image". A big flash image or a big binary lv_image source. If it's about the `lv_img` then have you used the image converter for v5.2: http://www.test.littlevgl.com/image-to-c-array

And do you use the `dev-5.2` branch from `lv_examples`?

mianhuacar, 2018-09-25:
Bin image mean a big binary lv_image source coverter by the image converter for v5.2: http://www.test.littlevgl.com/image-to-c-array. It will halt at read the file from the spi-flash.
Yes, I use the dev-5.2 branch from lv_examples. The problem seem to be like example lv_apps/demo dev5.2. Very strange,may be someting conflict in my code.

kisvegabor, 2018-09-25:
And is it working if you use only the demo example without any extra code?

mianhuacar, 2018-10-04:
Hi,
I change LV_MEM_CUSTOM  = 0, and the problem solved, so happy :)

closed 2018-10-04 by mianhuacar