lvgl/lvgl #428: Text input mask

mharizanov, 2018-09-26:
Hello,
I was wondering if there is some way to define text input masks for text area e.g. for text entry of IP address, phone #, credit card number, ZIP code, date, time etc?
Cheers

kisvegabor, 2018-09-26:
Hi,

In `dev-5.2` branch you can set the max length of text and the list of the accepted characters.
https://github.com/littlevgl/lvgl/blob/dev-5.2/lv_objx/lv_ta.h#L177

However, if you need something special, (e.g. in the case of IP addresses max `3 numbers` + `.`  + ...) you need to accept/reject the character one-by-one before adding them to Text area.

If you are using the Text area with lv_kb (keybord) and connected them with `lv_kb_set_ta` then you don't have an interface to accept/reject characters because every character is added internally by the library. If this is the case I can add an option to the text area to register a callback for this purpose. Maybe it would be more flexible than the `accepted character list` because this way it would be easy to extend the text area to support formats like you mentioned (date, time, etc).

For example:

```c
char * ta_ip_formatter(lv_obj_t * ta, const char * to_add);
```

In this function, you can check the input data and return anything based on it. For example, in the case of IP addresses,  after the third number, you can concatenate a '.'. You can execute further actions as well, like when the first number is deleted in a "segment", delete the dot as well by calling `lv_ta_del(ta)` or move the cursor as you need.

What do you think?

mharizanov, 2018-09-26:
Thanks for the input Gabor, yes I use it with a lv_kb object. 

IP address (for example) are a complicated case; the address bytes may be 1 to 3 characters, and preferably leading zeroes are not to be shown. For example "127.000.000.001" does not look good, I'd rather prefer "127.0.0.1". Solving this with text callback will probably be complicated?

I was thinking something like this:
[image]
The cursor will transit automatically to the next linked text area, if three digits are entered, a dot is entered via the keyboard, or focus moved.

Might be of interest as a solution to the problem, input text mask using control symbols:
https://support.office.com/en-us/article/control-data-entry-formats-with-input-masks-e125997a-7791-49e5-8672-4a47832de8da

kisvegabor, 2018-09-26:
I made some experimental tests with the callback of Text area I mentioned but it really seemed difficult to manage the content this way.

To transmit the cursor to the next block it seems you rather need a callback to the keyboard. It can be much simpler than the Text area's callback. It can be just a `validate` function which is called after every key hit. 

However, as the content type is specific to the Text area and not the Keyboard it should be somehow handled on the Text area's side. Your proposed solution about "input text mask" can work but it seems a bit complex and I can't add it right now. :( If you would like to implement it I kindly help along the way.  

As a workaround, I can imagine creating an `lv_task` (a function which is called periodically) to periodically check the content of the Text areas and do the required actions.

mharizanov, 2018-09-27:
Yes, I suspected that this won't be trivial to achieve, as it is more complex when you think about all the aspects. I'll research a bit more on "input text masks" and chime in if I come up with something.

Closing the question for now.

Thanks!

closed 2018-09-27 by mharizanov