lvgl/lvgl #429: Builtin symbols collide with adding custom symbols

poelstra, 2018-09-26:
I wanted to add more FontAwesome symbols, so I did something along the lines of:

```c
#define SYMBOL_GIFT            "\xEF\x81\xAB"

    // Add FontAwesome v900 10px to standard font
    lv_font_add(&fa_solid_900_10, &lv_font_dejavu_10);

    static lv_style_t style;
    lv_style_copy(&style, lv_obj_get_style(lv_scr_act()));
    //style.text.font = &fa_solid_900_10; // It does work fine if I uncomment this...

    lv_obj_t *icon = lv_label_create(lv_scr_act(), NULL);
    lv_label_set_style(icon, &style);
    lv_label_set_text(icon, SYMBOL_GIFT);
```

To get the same convenience of being able to mix normal text with icons (note: the style stuff is only to enable the 'workaround').

As a first test, I simply created `fa_solid_900_10.c` using your awesome online font converter, with the full range of characters (61440..62190). Then later I expected to be able to strip that down to just the icons I need in addition to the icons you already provided (I also enabled the symbol fonts in `lv_conf.h`).

This code crashes, however, because it tries to look up the bitmap in the symbol font, and thinks that that font includes the full range, it selects a non-existing character out of it, and crashes.

I suppose the built-in symbol fonts could either be converted to non-contiguous fonts (in which case it will 'fall-through' to other fonts that may provide the missing chars, but it's slower), or at least the accepted range should be 'clipped' to the characters actually in the set

In the latter case, probably also good to explicitly mention in the docs that the builtin symbol font should then be disabled (if you want to have the same nice mix of text and symbols).

Now, I'm not sure if this is somewhat 'by design', but it surely got me by surprise :)

kisvegabor, 2018-09-28:
Hi,

It's really a problem which I didn't consider so far. So the problem is the Symbols are stored continuously but the range is too big. 

We can't sore the symbols as sparse fonts because if `LV_TXT_UTF8 == 0` the range for the fonts is very limited. So storing then as continuous with the correct range seems suitable.

Anyway, the symbols are special. So I think the best would be to have a checkbox in the convert tool to mark that you are building a built-in font.

kisvegabor, 2018-10-01:
I think different mapping for UTF-8 and not UTF-8 might be confusing. 
If the range would be set correctly, it would be easy to add your own fonts as you need them. 
Maybe the checkbox is not required because the converter can check the range instead. If it's in the symbols' range then it can automatically add required modifications.

So, in summary, I find more simple if just the range would set correctly and you can add your font as you need them. What do think? Can it work?

kisvegabor, 2018-10-01:
Hi,

Meanwhile, I did some experiments:
1. Set the correct range for built-in symbols
2. Moved the built-in symbol fonts to begin at 0xF800 to reduce the probability of collision with fonts added by the user. (Fontawesome fonts begins at 0xF000)

I pushed the updates to `dev-5.2` Can you test how it works now?