lvgl/lvgl #431: lv_task_handler() hangs on first call - Porting to Teensy 3.6 & Adafruit 2.4" LCD

PalantirWielder, 2018-09-27:
Greetings,

Since discovering the existence of LittlevGL I've been working to port it to my project. I'm using the Arduino IDE with a Teensy 3.6 & Adafruit 2.4" LCD touchscreen. I've been using the [lvgl_test.ino](https://github.com/littlevgl/lv_projects/blob/master/arduino_library/lvgl_test/lvgl_test.ino) template in order to get things working, but haven't managed success quite yet.

My display's startup code works without issue, printing the debugging confirmations of setup completion to the screen. However as soon as loop() starts, lv_task_handler() is called but never returns. I know this because on the serial output I only get this debug output:
> 0: Start Task

I have tried my code with and without the debug printing with the same result.

If anyone could lend me some assistance in working through this issue, I'd appreciate it 😀

**More details:**
I'm using the Arduino IDE version 1.8.5, and commit 5c1a1d6e944728d093442da8f48dc0ea863e5fc8 from "dev-5.2" branch

I can post a copy of my lv_conf.h file if that'd be of assistance.

Here's my edited code for `lvgl_test.ino`:

`#include <SPI.h>
#include "ILI9341_t3.h"

#include <lvgl.h>

// ------------------ PINS --------------------
/* LCD Display */
#define TFT_CS   10  // Chip Select
#define TFT_DC    9  // Data/Command
#define TFT_LITE  8  // Backlight brightness control - PWM Output
#define TFT_RESET 7  // Reset LCD - Output

ILI9341_t3 tft = ILI9341_t3(TFT_CS, TFT_DC, TFT_RESET);  // Use hardware SPI (on Uno, #13, #12, #11) and the above for CS/DC

void my_disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t *color_array)
{

  /*TODO copy 'color_array' to the specifed coordinates of your display.*/
//  Serial.println("disp_flush(%d, %d, %d, %d, buffer) called");
  tft.writeRect(x1,y2, 1+x2-x1, 1+y2-y1, (const uint16_t *) color_array);
  
  /*Tell the flushing is ready*/
  lv_flush_ready();
}

bool my_tp_read(lv_indev_data_t *data)
{
  bool tp_is_pressed = false; /*TODO read here the state of touch pad*/
  int16_t last_x = 0;
  int16_t last_y = 0;
  
  if (tp_is_pressed)
  {
    /*Touch pad is being pressed now*/
    last_x = 0;       /*TODO save the current X coordinate*/
    last_y = 0;       /*TODO save the current Y coordinate*/  
  }
  
  data->point.x = last_x;
  data->point.y = last_y;
  data->state = tp_is_pressed ? LV_INDEV_STATE_PR : LV_INDEV_STATE_REL;

  return false;       /*Return false because no more to be read*/
}

void hal_init(void)
{
    /*Initialize the display*/
    lv_disp_drv_t disp_drv;
    disp_drv.disp_flush = my_disp_flush;
    lv_disp_drv_register(&disp_drv);

    
    /*Initialize the touch pad*/
    lv_indev_drv_t indev_drv;
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.read = my_tp_read;
    lv_indev_drv_register(&indev_drv);

    /*Initialize the graphics library's tick*/
    /*In a Timer call 'lv_tick_inc(1)' in every milliseconds, or if no other option just call it loop() */
}

void setup()
{
   // Set display pins before startup move SCK signal
  CORE_PIN13_CONFIG = PORT_PCR_MUX(0);
  CORE_PIN14_CONFIG = PORT_PCR_DSE | PORT_PCR_MUX(2);
  
  tft.begin();

  CORE_PIN13_CONFIG = PORT_PCR_MUX(0);
  CORE_PIN14_CONFIG = PORT_PCR_DSE | PORT_PCR_MUX(2);
  pinMode(13, OUTPUT);

  tft.fillScreen(ILI9341_BLUE);
  
  int16_t width_px = tft.width(), height_px = tft.height();
  float x_frac, y_frac, rad_frac;
  uint16_t screen_buffer[height_px][width_px];
  
  // Draw color gradients with writeRect() to confirm functionality
  for (uint16_t x = 0; x < width_px; x++)
  {
    x_frac = (float)x / width_px;
    
    for (uint16_t y = 0; y < height_px; y++)
    {
      y_frac = (float)y/height_px;
      rad_frac = sqrt(pow(x_frac,2) + pow(y_frac,2));
      screen_buffer[y][x] = tft.color565(255 * x_frac, 255 * y_frac, 255 * (1-(rad_frac / 1.414)));
    }
  }
  
  tft.writeRect(0,0, width_px, height_px, (const uint16_t *) screen_buffer);

  tft.setCursor(0, 0);
  tft.setTextColor(ILI9341_WHITE);  tft.setTextSize(1);
  tft.println("Hello World!");

  
  lv_init();
  tft.println("lv_init() complete");

  hal_init();
  tft.println("hal_init() complete");

  lv_obj_t *label = lv_label_create(lv_scr_act(), NULL);
  lv_label_set_text(label, "Hello Arduino!");
  lv_obj_align(label, NULL, LV_ALIGN_CENTER, 0, 0);
  tft.println("prep stuff complete. Exiting Setup");
}

void loop()
{
  static unsigned int counter = 0;
  // put your main code here, to run repeatedly:
  Serial.printf("%d: Start Task", counter);
  lv_task_handler();
  Serial.println(" - Finished");
  /*If 'lv_tick_inc(5)' is not called in a Timer then call it here*/
  Serial.print("Start Tick");
  lv_tick_inc(5);
  Serial.println(" - Finished");
  delay(5);             /*Wait a little*/
}`

PalantirWielder, 2018-09-27:
Here's the [lv_conf.h file](https://pastebin.com/dYBAnddF) that I'm currently using

PalantirWielder, 2018-09-27:
Also worth noting - for the following configuration the setup() code runs without issue:
```
#define USE_LV_LOG      1   /*Enable/disable the log module*/
#if USE_LV_LOG
#define LV_LOG_LEVEL    LV_LOG_LEVEL_WARN
#define LV_LOG_PRINTF   1
```

However if I drop `LV_LOG_LEVEL` to TRACE or WARN, then lv_init() enters but never returns. Additionally the callback function is never called.

If I change LV_LOG_PRINTF to 0, then lv_init() completes without issue on any logging level.

PalantirWielder, 2018-09-27:
I enabled the logging functionality and am code diving to find where the code gets stuck. I've currently chased the sticking code down to:
```
File: lv_refr.h
Function: static void lv_refr_obj(lv_obj_t * obj, const lv_area_t * mask_ori_p)
Line of where sticking is occurring: obj->design_func(obj, &obj_ext_mask, LV_DESIGN_DRAW_MAIN); 
```

Figured I'd mention this in case this rings any bells with anyone. If not I'll eventually dig down to the culprit's level.

PalantirWielder, 2018-09-27:
There was one random occasion yesterday when I actually had the "Hello Arduino!" text show up on the screen, albeit there were multiple copies of it vertically overlapping to the bottom of the screen.

At one point today it appeared that the sticking problem was gone. However as soon as I reprogrammed the Teensy with the same code it started sticking again.

I ended up disabling SHADOW, GPU, REAL_DRAW, and FILESYSTEM since less features could be preferable during this debugging process. Didn't seem to make a difference to the sticking.

Then on a whim I changed the optimization from "Faster" to “Fastest + Pure Code with LTO” which appeared to fix the sticking issue. However a misstep on my part prevented me from testing it further... I'm currently working to resolve that misstep.

So, with that said, since the issue sporadically disappears, and the sticking behavior seems sensitive to the compiler optimization, I'm guessing there's some sort of timing/memory-based issue.

PalantirWielder, 2018-09-28:
Well it seems that the LTO optimization causes issues with the Adafruit STMPE610 (I2C mode) touchscreen controller. While I haven't nailed a specific source of a problem, the effect is that screen presses are notably delayed. It appears that the touch.readData() calls take significantly longer when compiled with LTO. I verified that this effect also occurs with the simple STMPE610 example code, even with the printf() functions removed.

At this point the touchscreen isn't reasonably usable due to the latency, etc.

What can I do to either:
- have the LittlevGL compile in a non-LTO optimization level
or
- resolve the touchscreen I2C lag issue when compiling with LTO

I would opt for the former since in my online reading it appears that LTO can result in unwanted side-effects.

(Even though I read that LTO stands for Link-Time Optimization, I can't help but think of sandwich toppings :-P )

epikao, 2018-09-28:
hmm does I have maybe the same issue? see:
https://github.com/littlevgl/lvgl/issues/430

PalantirWielder, 2018-09-28:
What commit/release of the lvgl library are you using?

epikao, 2018-09-28:
version 5.2, I will try it with 5.1

PalantirWielder, 2018-09-28:
Specifically I'm using v5.1 [bd0c6c9ce3c38423b3825a657f0d9daf4244db55] and haven't implemented any dynamic code yet.

I can understand why interrupt-based object creation might not work. It's possible that your code is making changes to the LittlevGL system while it's in the middle of processing things. That is unless the library is "thread-safe" (I think that's the term) where it allows mid-processing changes. The author would have to speak to whether the code supports that or not.

If that's the case, then two ideas come to mind:
- Disable the updating interrupt when the Task/Tick functions are being processed
- Have the Task/Tick occur only in interrupt(s) with "higher" priorities than your updating interrupt. That would guarantee that Task/Tick cannot be running when your other interrupt takes control and starts making changes.
Just a thought

PalantirWielder, 2018-09-28:
It also appears that v5.1.1 [f5d4bf8fc47a3fc7fdca4b29cf156fa768084643] works as well for me.

PalantirWielder, 2018-09-28:
@epikao It looks like LV functions & interrupts don't mix well (Issue #416 )
[Partial quote:](https://github.com/littlevgl/lvgl/issues/416#issuecomment-424483169)
> It just came to mind that calling LittlevGL functions from interrupt can be problematic, because if you call `lv_task_handler` in the main `while(1)` the functions calls from the interrupt can mess up things if `lv_task_handler` makes something critical.

epikao, 2018-09-28:
hm, you ca  find my whole code here, the last post: https://github.com/littlevgl/lvgl/issues/430

so therefore I should call the  "lv_task_handler" with a timer too?
And in the tick / Timer function maybe I should set a varible to know if it's called or not... hmm I will try that asap

epikao, 2018-09-28:
>The better solution would be to just temporarily disable your object creation interrupt before the Task/Tick functions, then re-enable after they complete.

ok, that should be better yes... and with the lv_task_handler I suppose should be handled the same. Don't know if teensy knows interrup prioritys, need to check.

thank you

PalantirWielder, 2018-09-28:
K66P144M180SF5RMV2.pdf
**Section 3.2 - Interrupt priority levels**: This device supports 16 priority levels for interrupts.

So at the chip level it appears there are multiple priorities. Not sure if they're changeable, nor how they behave if interrupts of the same priority trigger at nearly the same time.

kisvegabor, 2018-09-28:
Hi,

Sorry, I couldn't answer yesterday but I followed the discussion.  

@PalantirWielder 
Thank you for figuring out these very valuable thinks. I tried to compile the code with lto on PC but "unfortunately" it worked. You said the program halts on `lv_refr_obj`. A few things came to my mind:
1. It's a recursive function. Is it possible that somehow it is not handled properly? 
2. Can you check the value of the following variables when the crash happens?  
- obj->design_func: just to be sure it's not NULL
- obj: just to be sure it's not NULL
- obj_ext_mask: all the 4 coordinates
3. Will the crash disappear if you immediately return  from `lv_refr_area_part_vdb` (lv_refr.c)?
4. What happens if you set `LV_MEM_SIZE   (16*1024)`. Maybe you got a stack overflow.
5. If `printf` is not supported in your system you need to register a print function the see the log of LittlevGL. You can use this function to do it: `lv_log_register_print`. In this case set `LV_LOG_PRINTF  0`

@epikao 
LittelvGL is not thread-safe by default so if you run `lv_task_handler` in the loop then you can't create objects in interrupts. I agree with PalantirWielder, you should disable while `lv_task_handler` is running.

PalantirWielder, 2018-09-28:
@kisvegabor 
The issue for me was that dev-5.2 _only_ would compile and not hang if it had LTO enabled. If I tried to compile _without_ LTO, then the code would get lost somewhere and never return.

1. Can't say I know enough about compiler optimizations 😑

2. I'd be glad to look into these. How do you suggest checking those values? (#5) I did find your logging system and registered my own Serial.printf() based callback for it, which is how I was tracing partially where the code managed to get stuck.

3. & 4. I will check and let you know.

The good news is that I know the LTO "requirement" somehow occurred between v5.1.1 and dev-5.2. There's probably a lot in there, but at least it limits the scope of where the issue(s) are hiding.

PalantirWielder, 2018-09-28:
Okay. I didn't think that the context of "Serial" would be available inside the lvgl library. I had assumed that I'd need to use the logging functionality somehow... I may end up writing or generalizing the logging function to allow a printf callback to be registered in my own source.

kisvegabor, 2018-09-28:
Oh, really I forget that it's not simple in Arduino.  
You can also do this:
```c
char buf[256];
sprintf(buf, design_func: %d", obj->design_func);
LV_LOG_TRACE(buf);
```

PalantirWielder, 2018-09-28:
Ah, yes. A much simpler solution 👍🏼

embeddedt, 2018-11-09:
@kisvegabor There hasn't been any activity here for a while. Should I close this issue?

PalantirWielder, 2018-11-09:
I haven't had my Teensy hardware available as of late so I haven't been able to push this forward. @kisvegabor Perhaps it'd be better to close this and I can open a new one with the latest release if I continue to run into issues?

kisvegabor, 2018-11-10:
@PalantirWielder 
Thanks for the feedback! I hope your issue it is solved in the new release.

closed 2018-11-10 by kisvegabor

Pablo2048, 2019-02-15:
Hi @kisvegabor - thanks for this perfect library. Unfortunately it seems like I hit the same wall. I'm trying to compile Arduino example for the ESP32 and same problem here. I turned on trace output and put some another into lv_refr.c and end on line `obj->design_func(obj, &obj_ext_mask, LV_DESIGN_DRAW_MAIN);` at which CPU ends with:

kisvegabor, 2019-02-21:
Probably the same issue was here too: https://github.com/littlevgl/arduino/issues/2#issuecomment-466004502