lvgl/lvgl #432: Freeze touch?

Masi2017, 2018-09-28:
Hi All,
I was wondering if there is any way to freeze touch on a specific object ? for example there is a message box with tow options and I want to make user select one of the options otherwise can't select other objects on the screen.

PalantirWielder, 2018-09-28:
Perhaps you're looking for something similar to what's discussed in issue #401?

kisvegabor, 2018-09-28:
Have a look at this: https://github.com/littlevgl/lvgl/issues/383

Masi2017, 2018-10-01:
Thanks All. I have both hardware keypad and touch, I can't disable click since my application has to work with both hardware keypad and touch pad. If I disable click , I can't select it using hardware keypad, am I right?

Masi2017, 2018-10-02:
@kisvegabor I have created a list , I added an action to each btn in the list, when user scrolling on the list, btns is clicked and action is run while I don't want that. I need the action be run ( btn call back function) only if user pressed that btn and not scrolling on the screen. Do you have any idea or maybe I'm doing something wrong.

kisvegabor, 2018-10-03:
For "freeze touch" you can use `lv_group_focus_freeze` the prevent focusing to an other object.

The buttons shouldn't call their action when they was scrolled. Please test in the PC simulator. Maybe something is wrong with your input device driver. If it's still not working please open a new issue for it.

kisvegabor, 2018-10-10:
> Thank you I did that but still I was able to select another object using touch and it worked only for hardware keypad.

Have you checked this?  #383 

> Would you please let me know if long press action works only for touch input or it works for hardware key as well?

Noe only lv_btn uses the long press but it should call the long press action when you press the LV_GROUP_KEY_ENTER long.

Masi2017, 2018-10-16:
@kisvegabor Thank you.

kisvegabor, 2018-11-02:
As the issue is solved I close this issue.

closed 2018-11-02 by kisvegabor