lvgl/lvgl #434: ddlist is not opened

Masi2017, 2018-09-28:
Hi All,
I need a "Select" button and once it is pressed I need to display a list of texts. User should be able to scroll among the options (texts) and  select one option and then a confirmation box will be appeared and if user confirmed , text should be sent. To do this I created a button, a ddlist to display my list and an mbox to get confirmation. I have both hardware keypad and touch pad. I need to open the ddlist once button is pressed but once user selects the button using touchpad, list is closed while I selected lv_ddlist_open(ddlist,true),   but it works fine with hardware keypad. Here is my function to create ddlist ( I tried to use Roller but I came across to other issue #433 ) :

lv_obj_t * DisplayddList(lv_obj_t * parent,rep_list_t *findOptions,lv_group_t *group,uint8_t VerMargin) {

  char *optionPtr=NULL;

  if (findOptions->func!=NULL)  optionPtr=(char *)findOptions->func(findOptions->data);
  else  optionPtr=findOptions->data;

  lv_obj_t *ddlist=lv_ddlist_create(parent, NULL);
  lv_ddlist_open(ddlist,true);

  uint8_t rowNum=calcFrequencyOfChar(optionPtr,'\n');

  lv_ddlist_set_options(ddlist,optionPtr);

  if (findOptions->func!=NULL)  free(optionPtr);

  lv_obj_align(ddlist, NULL, LV_ALIGN_IN_TOP_LEFT, 10, VerMargin);

  if (rowNum>7)
  lv_ddlist_set_fix_height(ddlist,lv_obj_get_height(parent)-20);//LV_VER_RES-80);

  lv_ddlist_set_hor_fit(ddlist, false);
  lv_obj_set_width(ddlist,lv_obj_get_width(parent)-20);//

  lv_ddlist_set_action(ddlist, ddlist_action);  /*Set a function to call when anew option is chosen*/

  lv_ddlist_set_sb_mode(ddlist, LV_SB_MODE_AUTO);

  if (group!=NULL)
  {
    lv_group_add_obj(group,ddlist);
    lv_group_focus_obj(ddlist);
  }

  return ddlist;

}

kisvegabor, 2018-09-28:
Hi,

`lv_ddlist_open(ddlist,true);` opens the ddlist once and not keeps it open after presser or defocus. If you something which remains open you should use a simple "lv_list".

kisvegabor, 2018-10-07:
As this is the intended behavior of the Drop down list I close this issue.

closed 2018-10-07 by kisvegabor

Masi2017, 2018-10-16:
I noticed that ddlist object is opened whenever is focused , is there anyway to open it only if user press Enter on that or touch it?

kisvegabor, 2018-10-17:
There is no option for this now. However, it's possible improvement. If you'd like to add it I can help with the implementation.

Masi2017, 2018-10-17:
Ok that would be great! Any help would be highly appreciated.