lvgl/lvgl #435: Touch Press-Release Audio Feedback

PalantirWielder, 2018-09-28:
Before I found LittlevGL I was in the process of writing my own primitive GUI for my touchscreen. In it I had code that would play a "tick" noise each time one of my buttons registered a press or depress/release.

If I only cared if I was touching _somewhere_ on the screen then I'd put the clicking functionality in the input read callback. However I'd like the noise generation to exclusively be when an object has actually been triggered/released.

Is there a general callback that is called when any active object is pressed/released/held? Something like that would be preferable over having to add the click callback to every active object that I add.

kisvegabor, 2018-09-28:
You can do it with a little trick. In `lv_obj.c` in `lv_obj_signal` function add:
```c
else if(sign ==  LV_SIGNAL_PRESSED) {
        printf("Clicked\n");       //Or start to play the click sound
    }
```

It will be called every time when an object is pressed (not only `lv_obj`, `lv_btn`, `lv_ddlist` etc as well).

PalantirWielder, 2018-09-28:
Perfect. Thanks!

If that were built out so a callback could be registered, would you be interested in my submitting the change as a PR?

And if so, do you have any requirements/suggestions?

kisvegabor, 2018-09-29:
Yes, a PR would be welcome! :)

First of all, we should find a good name for this feature. My idea is `feedback`. 
I was thinking about maybe this feature should be connected to the input devices. Because it is possible that you want different feedback for a touch pad and an encoder.

So I can imagine an interface like `lv_indev_set_feedback_cb(indev, my_feedback)` where `my_feedback` has a bool parameter. true: an object activated; false: an object is deactivated.
In this case, the saved feedback function should be called with true with 
- `LV_SIGNAL_PRESSED`
- `LV_SIGNAL_FOCUS`
and false with:
- `LV_SIGNAL_PRESS_LOST`
- `LV_SIGNAL_RELEASED`
- `LV_SIGNAL_DEFOCUSE``

FYI, in `lv_obj_signal` the current input device can be got with `lv_inedv_get_act()`. (Can return NULL)

What do you think?

PalantirWielder, 2018-10-01:
Hmm... feedback could work.

> it is possible that you want different feedback for a touch pad and an encoder.

Good point. For a turning knob it might be nice to have it make a clicking noise, but perhaps not with another input.

While I haven't learned enough about your software to know it already, I feel that there might be instances where developers would want know the difference between a "pressed" event and a "focus" event". Is it possible that we could simply pass one of the six listed "LV_SIGNAL_" enums to the callback function if one of them is found, and if a callback has been registered? That way any processing could be done on the callback's side.

kisvegabor, 2018-10-01:
I tried to add the delimiters to the signals but I realized that there are some signals which are not relevant in this case. For example, there is nothing to do with:
- LV_SIGNAL_DRAG_BEGIN
- LV_SIGNAL_DRAG_END
- LV_SIGNAL_GROUP_GET_EDITABLE

In addition `LV_SIGNAL_LONG_PRESS` is used only by `lv_btn`. This signal will have effect only if a long press action is set for a button. So it makes sense to me to call the feedback function by in the `lv_button_signal` if something happened on long press. 
 
So I vote for choosing the signals manually. What do you think?

PalantirWielder, 2018-10-01:
I was thinking that the delimiters would just be used to determine whether or not the feedback callback would be triggered or not. So instead of having an IF statement with all the Input related and Group related signals, we could just check to see if the signal was between the two sets of delimiters:
((Input related start < signal) && (signal < input related end)) || ((group related start < signal) && (signal < group related end))
Then it would pass the specific signal to the callback, and inside the callback it would choose whether to respond to the signal or not.
That way any specific signal processing is on the developer's side, while your code only checks for:
Is there a callback registered?
Is the signal between the Input related delimiters, or the group related delimiters?

This would in my mind simplify the code changes to your library, and offload any special programing to the developer to access any of those two groups of signals an do whatever they'd like to in their callback.

So while it might send "irrelevant" signals to the callback, the callback would decide whether to ignore or handle the signals at the developer's discretion. It's possible there are developer's that might find applications for receiving the otherwise "irrelevant" signals, and this would allow them the freedom to use them, instead of constraining the interface.

What do you think? Would there be a problem sending the "irrelevant" signals if the callback decided whether or not to respond to them?

kisvegabor, 2018-10-01:
It totally makes sense but: 
-  `LV_SIGNAL_LONG_PRESS` will be sent if the object is pressed for enough time. But only the Button object knows whether something happened on long press or not (is there long press action callback registered or not). And giving feedback on long press if nothing happens might be confusing.  
- It's the same for `LV_SIGNAL_LONG_PRESS_REP` which makes send only for the Keyboard and button matrix object. It would be ugly to send feedback repeatedly when nothing happens.
 
So I say use as many signals as we can in `lv_obj_signal` but some of them need special handling. That's why it seems to me we can't pass all the indev and group related signals is `lv_obj_signal`.

PalantirWielder, 2018-10-01:
Sounds good. I'll follow up once I have the changes implemented, tested, and ready.

Also, which branch/release/commit would you prefer I build my PR on? As we've discussed I cannot compile and run dev-5.2, but I can use v5.1.1.

turoksama, 2018-10-02:
And vibration...

PalantirWielder, 2018-10-02:
@turoksama Thankfully with the future ability to register a callback, the custom function you write could do anything from playing a sound, activating a Taptic vibration, or light up an LED. If you can activate the idea from a function, you'll be able to tie it to this future feature 😀

PalantirWielder, 2018-10-03:
@kisvegabor 
1. From the following, which signals _should_ I include?
```
/ *Input device related */
LV_SIGNAL_PRESSED,
LV_SIGNAL_PRESSING,
LV_SIGNAL_PRESS_LOST,
LV_SIGNAL_RELEASED,
LV_SIGNAL_LONG_PRESS,
LV_SIGNAL_LONG_PRESS_REP,
LV_SIGNAL_DRAG_BEGIN,
LV_SIGNAL_DRAG_END,

/* Group related */
LV_SIGNAL_FOCUS,
LV_SIGNAL_DEFOCUS,
LV_SIGNAL_CONTROLL,}
```

2. If necessary, would it be acceptable change the order of the items in each group such that the signals we report are adjacent? And could I insert delimited listings to create testable bounds? That way I could just test for ENUMs between the delimited ENUMs instead of testing for each included signal individually.

3. You mentioned the signal `LV_SIGNAL_GROUP_GET_EDITABLE`. Where is that from? I don't see it in the v5.1.1 list in lv_obj.h.

kisvegabor, 2018-10-03:
Hi,

I think this signals should be added to `lv_obj_signal`:
- LV_SIGNAL_PRESSED
- LV_SIGNAL_PRESS_LOST
- LV_SIGNAL_RELEASED
- LV_SIGNAL_DRAG_BEGIN
- LV_SIGNAL_DRAG_END
- LV_SIGNAL_FOCUS
- LV_SIGNAL_DEFOCUS
- LV_SIGNAL_CONTROLL
I think we can't use the delimiter

`LV_SIGNAL_GROUP_GET_EDITABLE` is in `dev-5.2`. It fine if you add these to v5.1.1 and I will merge it `dev-5.2`. I will add the `LV_SIGNAL_LONG_PRESS` and `LV_SIGNAL_LONG_PRESS_REP` because they are a little bit tricky. I will explain where to add these signal in v5.1.1 if you can't update to `dev-5.2` 

Because some signals need special handling (as discussed above)  I think it might be confusing to add delimiters which work only in `lv_obj_signal`. So I would prefer to handle the signals one-by-one. If new signals come they need to be reviewed in this regard anyway. 

If this feature will be added I will release v5.2! :)
(However, I know it's not so important for you. :( )

PalantirWielder, 2018-10-03:
Sounds good. I'll add those signals individually then. I didn't want to have a bunch of if-statements the processor would have to check through during each polling, but if it's better for the code structure then I'll conform my code accordingly.

And yeah, since I cannot use v5.2 at the moment, I'll make the changes to v5.1.1 and submit my PR once I'm done with my current project.

kisvegabor, 2018-10-05:
Great, thank you! :)

I'd like to release v5.2 this week so it seems this feature will go into the next release instead.

PalantirWielder, 2018-10-05:
No worries. The timing didn't line up this release, but the feature will make it in eventually.

kisvegabor, 2018-12-18:
@embeddedt started to implement it here: https://github.com/littlevgl/lvgl/pull/641

closed 2018-12-18 by kisvegabor