lvgl/lvgl #438: Bugs: checkbox & drawing etc.

turoksama, 2018-10-02:
Hi there,
     I modified a theme from "theme_material" and tested with "**lv_test_theme_1(th);**", some bugs appeared as follows:
     1. when setting **style.body.radius=0;** , drawing one dot more in every 90' angle as below:
[image]
I pointed out 2 for example. Both in "pc_simulator" & hardware board.
I tried to fix that but "xxx_design" is more complicated than what i thought to be.
Use an "if" condition before drawing may higher efficiency?
     2.knob in slider: a square will be drawn when **style.body.radius=0;**
I modified these lines in "lv_slider_design()" : line 327-328
if(slider_w >= slider_h) {
            if(ext->knob_in == 0) {
                knob_area.x1 = area_indic.x2 - slider_h / **4**;
                knob_area.x2 = knob_area.x1 + slider_h**/2**;
            } else {
                knob_area.x1 = (int32_t) ((int32_t)(slider_w - slider_h) * (cur_value - min_value)) / (max_value - min_value);
                knob_area.x1 += slider->coords.x1;
                knob_area.x2 = knob_area.x1 + slider_h;
            }

            knob_area.y1 = slider->coords.y1;
            knob_area.y2 = slider->coords.y2;
        }
     3.when i try to merge to beta version, check box error in simulator: wrong size ,no caption and check will crash and have no effects after reverting back (ver 5.1.0 is fine before that), acts normal in hardware board(5.2.0 beta earlier version about 3-4 months ago)
     4. no effects on tab text font size:

     lv_style_t tb;

    lv_theme_set_current(th);
    lv_obj_t *scr = lv_cont_create(NULL, NULL);
    lv_scr_load(scr);
    lv_cont_set_style(scr, th->bg);

   tv = lv_tabview_create(scr, NULL);
   lv_style_copy(&tb, th->tabview.bg);
   tb.text.font = &muli_latin_regular_40;
   lv_tabview_set_style(tv, LV_TABVIEW_STYLE_BG, &tb);

[image]

[lv_theme_flat.zip](https://github.com/littlevgl/lvgl/files/2436228/lv_theme_flat.zip)

turoksama, 2018-10-02:
i don't know why the page looks so messy ~_~

kisvegabor, 2018-10-02:
Hi,

please put your code section  between 
` ```c`
code block
` ``` `
tags

turoksama, 2018-10-02:
@kisvegabor 
Hi, I found that ddlist & list & roller are much alike, commonly an downward arrow on the right by the ddlist, maybe more user friendly? What do you think?

kisvegabor, 2018-10-03:
Thank you for updating your comment!

1. radius: I fixed it in `dev-5.2` branch. Can you verify if it works well?
2. square knob in slider: yes, the knob can be only square (or circle with LV_RADIUS_CIRCLE radius)
3. crash: your style declaration should be: `static lv_style_t tb;` because only the style pointers are saved.
4. To set the text size on tabs you should set the styles like this:
```c
   /*The released button is transparent so it will inherit the background's style*/
   static lv_style_t tb_bg;
   lv_style_copy(&tb_bg, th->tabview.btn.bg);
   tb_bg.text.font = &muli_latin_regular_40;
   lv_tabview_set_style(tv, LV_TABVIEW_STYLE_BTN_BG, &tb_bg);

   /*The toggled released (active tab) is not transparent. Set it directly*/
   static lv_style_t tb_tgl_rel;
   lv_style_copy(&tb_tgl_rel, th->tabview.btn.tgl_rel);
   tb_tgl_rel.text.font = &muli_latin_regular_40;
   lv_tabview_set_style(tv, LV_TABVIEW_STYLE_BTN_TGL_REL, &tb_tgl_rel);
``` 

I pushed a minor fix to `dev-5.2` related to tabview styling.

turoksama, 2018-10-03:
@kisvegabor 
Thanks a lot for your advise.
First,
We have a major problem here: hardware board have a lot of glitches on the screen with 5.2.0 beta(new architecture on drawing) and is much slower than 5.2.0 beta earlier version. So, i am not sure which file should be replaced to solve the radius problem.
Second,
We have a new situation here:
[image]
Alignment of labels are not the same when using animation(the containers of labels using "lv_cont_set_layout(t_inf, LV_LAYOUT_ROW_M);" ) :
first one not using long mode.
second one using LV_LABEL_LONG_ROLL long mode.
I think the animation have bugs : perhaps layout param is not inherited?

kisvegabor, 2018-10-03:
Hi,

Please update to the `dev-5.2` branch and test with that.

kisvegabor, 2018-10-03:
If it"s not working please provide an example code to test your code.

turoksama, 2018-10-03:
Thanks for your advise.
We will looking forward to testing the new beta.

turoksama, 2018-10-04:
@kisvegabor 
Hi,
I update to the latest dev-5.2 branch and test with that, and make efforts to optimize for speed. Alignment of labels are not the same when using animation problem is still there.

[mframe.zip](https://github.com/littlevgl/lvgl/files/2444833/mframe.zip)
regards!

kisvegabor, 2018-10-05:
Hi,

I checked your code and modified one line to make it work:
```c
lv_obj_set_width(t_inf_label, INFO_LANE_LABEL_W);
```

Originally you set the label and the conatainer height to the same value. The label objcet was really cenered vertically  due to the layout. However, the text always begins at the top of the label.

See the difference:

Your version:
[image]

My version:
[image]

So I just let the height to set automatically when setting the text with `lv_label_set_text` and later only modifed the width.

turoksama, 2018-10-05:
Hi,
Your code works fine.Thanks a lot.
Regards!

closed 2018-10-05 by turoksama

kisvegabor, 2018-10-05:
Your welcome!