lvgl/lvgl #439: Selectively Disabling Tab Clicks

PalantirWielder, 2018-10-02:
The application that I'm developing will have three viewable panes/tabs.
When the system starts I want to start on Tab #<span>1</span>, with <span>#</span>2 & <span>#</span>3 disabled.
Then once criterion have been satisfied, I want to make <span>#</span>2 clickable as well.
Then a button within Tab #<span>2</span> will move the view to Tab <span>#</span>3 and disable 1&2.

Is this possible? If so, could someone point me towards info/examples that demonstrate this?

kisvegabor, 2018-10-03:
There is no APi for it but  can do it with a little trick: 
```c
    lv_tabview_ext_t * tv_ext = lv_obj_get_ext_attr(tv1);
    char ** btnm_map = (char**)lv_btnm_get_map(tv_ext->btns);       /*Get the button matrix's descriptor*/
    btnm_map[1][0] |= LV_BTNM_INACTIVE_MASK;                        /*Make the 2th button inactive by modifying the control byte*/
```

Note that, still you will be able to slide to the inactive tab.

If it works well, you can add API for it if you want!

kisvegabor, 2018-10-05:
You can disable sliding with `lv_tabview_set_sliding`. 

You can add tabs on the fly but there is no animations right now. What kind of animation do you imagine?

PalantirWielder, 2018-10-05:
Thanks to your documentation online I did find and employ that function.

Tangent: I believe the Container website documentation has Enums that are incorrect. If my memory serves me I copied one out and it had an unnecessary "_contain" or something like that they I had to remove.

I tried the various themes to see how the tabs looked, but for some reason the Materials theme init wasn't being found, but all the others were. And they were enabled in the conf file. Have you encountered that before?

I was thinking something fairly simple. If it were an addition on the edge I'd add the tab and resize the tabs bar so nothing visibly changed. Then I'd resize the bar so it fit the screen eventually showing the new tab. Removal from the edge could be the reverse.
An alternative would just be to animate the width of the tab so it shifted between full with and 0 width. Just off the top of my head that's what I thought.

PalantirWielder, 2018-10-05:
Found problem and solution to above mentioned Materials theme issue. Filed #444

kisvegabor, 2018-10-07:
Hi,

I fixed the `LAYOUT` enum names. Thank you!

Animations:
The possibilities to animate the bar are limited because the bar is a `Button matrix` object and you can set only the ration between the width of the buttons on it. You can use numbers between [1..7] to set the relative width of the buttons so an animation on it would be quite stepped.
However, an animation on the content page's width sounds well!

PalantirWielder, 2018-10-11:
When I went to compile the button disabling code, it wouldn't compile because:
`lv_page_ext_t * tv_ext = lv_obj_get_ext_attr(tabview);`
throws
> error: invalid conversion from 'void*' to 'lv_page_ext_t*' [-fpermissive]
>    lv_page_ext_t * tv_ext = lv_obj_get_ext_attr(tabview);

This seems like it shouldn't be difficult to resolve, like by casting the type from void * to the other, but I haven't managed to reason out what it should be. If the solution is obvious to you, I could use the assistance :-)

kisvegabor, 2018-10-12:
Great! :)