lvgl/lvgl #440: Where can i get the pc_simulator _5.2

mianhuacar, 2018-10-02:
Hi,

Where can i get the pc_simulator _5.2. I had tried changed pc_simulator _5.1 to pc_simulator _5.2 myself, but fail. After changed, it cannot show png image(convert by lv_utils-dev-5.2) correctly.
I should be
[image]
but
[image]
or
[image]

turoksama, 2018-10-03:
hooray! 5.2 is coming…

mianhuacar, 2018-10-04:
Hi,
I change #if LV_COLOR_DEPTH == 24 ->#if LV_COLOR_DEPTH == 32, and ok now :)

closed 2018-10-04 by mianhuacar