lvgl/lvgl #441: Support for an OLED driven by SSD1322

e-music, 2018-10-03:
Hi,

I was wondering if this library supports OLED modules with SSD1322 display controller. 

Thank you

e-music, 2018-10-03:
Hello,

Thank you very much for the prompt reply. Yes, I guess I still have the driver from last time used it with another GUI library. It has only basic functions to initialize, write command/data, etc... Anyways, if you can suggest a particular driver or refer me to one, I would really appreciate your help! 

I can't wait to test this on my OLED! 

Best regards

PalantirWielder, 2018-10-04:
@e-music 
Does the list of basic functions include a function to push an array of pixels to the screen?
For example some Arduino-based graphics drivers include a writeRect() function which takes in the starting (x,y) coordinates, the width and height to draw, and then an array of 2 byte elements that contain the RGB values for the pixel.

Do you see anything like that?

e-music, 2018-10-05:
@PalantirWielder 

I had a driver implemented inside another GUI lib designed for monochrome LCDs. However, the base driver I got off the manufacturer's website has the following:

`

void Write_Command(unsigned char command)
{
   PORTA=command;
   CLR_CS;
   CLR_CD;
	 SET_RD;
   CLR_WR;
		_delay_us(5);
   SET_WR;
		_delay_us(5);
   SET_CS;
}

void Write_Data(unsigned char date)
{
   PORTA=date;
   CLR_CS;
   SET_CD;
	 SET_RD;
   CLR_WR;
		_delay_us(5);
   SET_WR;
		_delay_us(5);
   SET_CS;
}

void Command_Lock(unsigned char lock)
{
   unsigned char value;
   value = lock;
   Write_Command(0xfd); 
   Write_Data(value); 
} 

void Sleep_Mode(unsigned char mode)
{
   unsigned char value;
   value = mode;
   Write_Command(value); 
}
void Column_Address(unsigned char start , unsigned char end)
{
   unsigned char value_1,value_2;
   value_1 = start ;
   value_2 = end ;
   Write_Command(0x15); 
   Write_Data(value_1); 
   Write_Data(value_2); 
}

void Row_Address(unsigned char start , unsigned char end)
{
   unsigned char value_1,value_2;
   value_1 = start ;
   value_2 = end ;
   Write_Command(0x75); 
   Write_Data(value_1); 
   Write_Data(value_2); 
}
void Dual_COM_Line(unsigned char mode_1, unsigned char mode_2)
{
   unsigned char value_1,value_2;
   value_1 = mode_1;
   value_2 = mode_2;
   Write_Command(0xA0);
   Write_Data(mode_1); 
   Write_Data(mode_2);
}
void Display_Start_Line(unsigned char line)
{
   unsigned char value;
   value = line;
   Write_Command(0xA1); 
   Write_Data(value); 
} 

void Display_Offset(unsigned char offset)
{
   unsigned char value;
   value = offset;
   Write_Command(0xA2); 
   Write_Data(value); 
} 
void Fun_Selection(unsigned char selection)
{
   unsigned char value;
   value = selection;
   Write_Command(0xAB); 
   Write_Data(value); 
} 

void Phase_Length(unsigned char length)
{
   unsigned char value;
   value = length;
   Write_Command(0xB1); 
   Write_Data(value); 
} 

void Clock_Divider(unsigned char divider)
{
   unsigned char value;
   value = divider;
   Write_Command(0xB3); 
   Write_Data(value); 
} 

void Display_En_A(unsigned char enhance_1, unsigned char enhance_2)
{
   unsigned char value_1,value_2;
   value_1 = enhance_1;
   value_2 = enhance_2;
   Write_Command(0xB4);
   Write_Data(enhance_1); 
   Write_Data(enhance_2);
}

void Set_GPIO(unsigned char gpio)
{
   unsigned char value;
   value = gpio;
   Write_Command(0xB5); 
   Write_Data(value); 
} 
void Sec_Pre_Period(unsigned char period)
{
   unsigned char value;
   value = period;
   Write_Command(0xB6); 
   Write_Data(value); 
} 

void Default_Gray(unsigned char gray)

{
   unsigned char value;
   value = gray;
   Write_Command(value); 
}

void Pre_Voltage(unsigned char voltage)
{
   unsigned char value;
   value = voltage;
   Write_Command(0xBB); 
   Write_Data(value); 
} 

void Set_Vcomh(unsigned char vcomh)
{
   unsigned char value;
   value = vcomh;
   Write_Command(0xBE); 
   Write_Data(value); 
} 
void Set_Contrast(unsigned char contrast)
{
   unsigned char value;
   value = contrast;
   Write_Command(0xC1); 
   Write_Data(value); 
} 

void Master_Contrast(unsigned char current)
{
   unsigned char value;
   value = current;
   Write_Command(0xC7); 
   Write_Data(value); 
} 
void MUX_Ratio(unsigned char ratio)
{
   unsigned char value;
   value = ratio;
   Write_Command(0xCA); 
   Write_Data(value); 
} 

void Display_En_B(unsigned char enhance_1, unsigned char enhance_2)
{
   unsigned char value_1,value_2;
   value_1 = enhance_1;
   value_2 = enhance_2;
   Write_Command(0xD1);
//   (enhance_1); 
   Write_Data(enhance_2);
}
void Display_Mode(unsigned char mode)
{
   unsigned char value;
   value = mode;
   Write_Command(value); 
}
void Reset_IC()
{
   _delay_ms(10);
   CLR_RESET;
   _delay_ms(1);
   SET_RESET;
}
void fill(void)
{
        unsigned char x,y;
        unsigned char dat=0;
   Column_Address(0x1C,0x5B);         //Set Column Address
   Row_Address(0x00,0x3F);            //Set Row Address
        Write_Command(0x5c);
       for(y=0;y<64;y++)
       {
           dat=0;
           for(x=0;x<64;x++)
           {
            Write_Data(dat);

            Write_Data(dat);
            if((x%8==0)&&(x>0))
               dat+=0x22;
            }
        }
}
void fill2(void)
{
        unsigned char x,y;
        unsigned char dat=0;
   Column_Address(0x1C,0x5B);         //Set Column Address
   Row_Address(0x00,0x3F);            //Set Row Address
        Write_Command(0x5c);
       for(y=0;y<64;y++)
       {
           dat=0xFF;
           for(x=64;x>0;x--)
           {
            Write_Data(dat);
            Write_Data(dat);
            if((x%8==0)&&(x>0))
               dat-=0x22;
            }
        }
}

`

PS. I have intentionally removed the initialize block

PalantirWielder, 2018-10-05:
If you can get that driver implemented on your current project, and can use fill() successfully, then that function could likely be copied and altered to do the job.

It looks like it fill the screen with a specific pixel value starting at a particular column and row location. At the very least it could be used to draw a row of pixels. If the screen allows you to specify an arbitrary "box" of pixels to fill that'd be even better for performance.

Once you have the position control implemented then you'll need the new function to draw pixels from the incoming pixel array instead of the constant data fill() currently sends. Shouldn't be too much work to bridge that last gap in the driver.

PalantirWielder, 2018-10-05:
Awesome! I'm glad to hear it wasn't difficult.

I believe antialiasing can be enabled/disabled as desired in the lv_conf.h file under LV_ANTIALIAS

C47D, 2018-10-05:
@e-music, do you have any pictures of your screen available for us to see them?

e-music, 2018-10-05:
Of course, I do. Do you want to see the LCD module itself or the displayed pictures?

e-music, 2018-10-05:
Here you go:
[image]
[image]

I hope that helps...

PalantirWielder, 2018-10-05:
I believe DPI is supposed to be Dots/Pixels Per Inch. So if you know one of the screen dimensions and it's pixel count you should be able to calculate the DPI.

I believe the DPI is used for setting the scaling of objects. So you should be able to tweak it to scale things differently.

C47D, 2018-10-05:
@e-music, thanks for the pictures, i haven't a lot of free time to run lvgl on my project but this make me want to.

PalantirWielder, 2018-10-05:
I calculated the same value 👍

kisvegabor, 2018-10-06:
I'm also curious about an image with LittlevGL!

e-music, 2018-10-06:
@PalantirWielder 
OK. I'm back now to LittlevGL. I setup a timer and in its interrupt handler I have added lv_tick_inc(1); and in the main while loop I have added lv_task_handler(); with a delay of 5 ms between each iteration. All I need now is not adjust one or more of the existing functions to support the lib or in other words, to associate it with the GUI lib.

e-music, 2018-10-06:
I was wondering if there is any example for these functions on a monochrome LCD:

static void ex_disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p);
static void ex_disp_map(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p);
static void ex_disp_fill(int32_t x1, int32_t y1, int32_t x2, int32_t y2,  lv_color_t color);

I guess **color_p** has to be on/off or a grayscale level in my situation, but how to handle that?

e-music, 2018-10-07:
@PalantirWielder 
Yes of course. It is a 4-bit per pixel and supports 16 grayscale levels.

e-music, 2018-10-07:
@kisvegabor 
With that being said, both of disp_fill and disp_map are not required when "buffered drawing" is used. Of course, I would rather prefer to get things done quickly so it's better if disp_flush does the job alone. 

I guess I need some time before I can figure it out on how to convert color to its corresponding gray-scale value (brightness level). 

It would be highly appreciated if you can supply some sample source. 

> The frame buffer is required if you can refresh only full bytes on your display but a byte can mean 2 pixels in case of 4-bit color depth. So to avoid reading back the unknown pixel if the for row/col is an odd value you should use the whole frame. I'm not sure if it is problem for you or not.

Does that mean I have to send an entire frame buffer to the LCD for every single pixel I want to draw?

Zaltora, 2018-10-08:
Nope, i use SSD1306. I will  update the driver with new feature from 5.2.

kisvegabor, 2018-10-08:
@Zaltora  I see, thank you!

e-music, 2018-10-08:
@kisvegabor 
Nice, I just realized that lv_color_brightness() is in fact a function. I thought you were referring to something else. I guess this has to be called before drawing to the LCD takes place. Now, I just need to figure out a way to convert Column Start/End address, Row Start/End address to (x, y) coordinates.

PalantirWielder, 2018-10-09:
It appears that [lv_color_brightness()](https://github.com/littlevgl/lvgl/blob/83cef11a78f64be10e0fe4bc7b8c05d281d3e438/lv_misc/lv_color.h#L357) takes in a pixel and spits out the brightness in 8-bits. So without having an entirely separate buffer to convert into, you may just want to take each pixel from pixel buffer being fed to _flush(), and run it thought lv_color_brightness() just before sending it to the LCD:

I would imagine it'd be something like:
```
ts_my_disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t *color_array)

// do stuff to prep display

// make loop through buffer
  // then to convert 2 pixels at a time:
  pix_4bit[0] = lv_color_brightness(color_array[ i ])  >> 4;  // Convert to grayscale then shift away lowest 4-bits
  pix_4bit[1] = lv_color_brightness(color_array[ i+1 ]) >> 4;
  buffer_byte_to_send = pix_4bit[0] || (pix_4bit[1] << 4); // set pix 'i' as lower nibble, and 'i+1' as upper nibble.
  // send buffer_byte_to_send to display
```

or if that is easily clear, you could replace the conversion with:
```
buffer_byte_to_send = (lv_color_brightness(color_array[ i+1 ]) & 0xF0) || (lv_color_brightness(color_array[ i ])  >> 4);
```
which may be slightly faster.

Anyway, that's just the pseudo-code for what I imagined. It may or may not be the avenue you'll need to follow.

kisvegabor, 2018-11-02:
@e-music Is there any news in this topic?