lvgl/lvgl #442: Handling LV_SIGNAL_PRESS_LOST

PalantirWielder, 2018-10-04:
I had originally planned to be able to press a button to activate a motor, then upon release of that button deactivate the motor. As such I added callbacks for LV_BTN_ACTION_PR & LV_BTN_ACTION_CLICK. 

However if I press a button and then slide my finger off of it, the button deactivates and the motor stays activated.

Is there something I can do to capture the button press being lost? I know in the input feedback function I have shoehorned in I can detect the LV_SIGNAL_PRESS_LOST, however that's without the context of the button, and not through the typical lv_btn_set_action() API.

I'd rather not depend upon the periodical signalling of a held button, nor would I prefer to hack together a solution that uses my input feedback function to catch the action.

Would it be possible to easily increase the action[] callback list to respond to the press being lost? or would the current architecture make that difficult/messy?

kisvegabor, 2018-10-05:
Hi,

You can extend the signal function of the button like this:
```c

/*Global declarations*/
static lv_signal_func_t btn_signal;
static lv_res_t my_btn_signal(lv_obj_t * btn, lv_signal_t sign, void * param);

...

    lv_obj_t * btn = lv_btn_create(lv_scr_act(), NULL);
    btn_signal = lv_obj_get_signal_func(btn);            /*Save the original signal function*/
    lv_obj_set_signal_func(btn, my_btn_signal);        /*Set a new signal function*/

...

static lv_res_t my_btn_signal(lv_obj_t * btn, lv_signal_t sign, void * param)
{
    lv_res_t res;

    /* Call the original signal function */
    res = btn_signal(btn, sign, param);
    if(res != LV_RES_OK) return res;

    /*Proces the signals*/
    if(sign == LV_SIGNAL_PRESS_LOST || sign == LV_SIGNAL_RELEASED) {
        printf("Press lost\n");
    }

    return res;
}
```

kisvegabor, 2018-10-09:
Is it working?

PalantirWielder, 2018-10-09:
I should hopefully find out today or tomorrow. I've been finishing up the back end code so the GUI actually has something to call. I'll let you know

PalantirWielder, 2018-10-11:
It does in fact work! Thanks :-D

closed 2018-10-11 by PalantirWielder

kisvegabor, 2018-10-12:
Your welcome!
As it's not supported directly right now this small "hack" should be applied. In v6.0 the actions will be modified the include the triggering event in the callback function. It will provide more flexibility and "PRESS_LOST" can be added as well.
More info here: https://github.com/littlevgl/lvgl/issues/316