lvgl/lvgl #443: problem in calling lv_font_get_bitmap()

ywkelvin, 2018-10-04:
Hi,

I would like to get the pointer value of the font by using the function lv_font_get_bitmap(&lv_font_dejavu_10, 0x20) directly called at main.cpp. It shows an error message as follows when I compile the code.

main.cpp:26:32: error: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]

Is there anything wrong ?

Best Regards,
Kelvin

ywkelvin, 2018-10-04:
Hi,

Can I know whether the problem can be fixed by revising the line at typedef struct _lv_font_struct {} in lv_font.h 
- const uint8_t  (*get_bitmap)(struct _lv_font_struct * ,uint32_t);    /*Get a glyph's  bitmap from a 
+ const uint8_t * (*get_bitmap)(struct _lv_font_struct * ,uint32_t);    /*Get a glyph's  bitmap from a 

Regards,
Kelvin

embeddedt, 2018-10-04:
LVGL font bitmaps are stored in `const` format. See https://raw.githubusercontent.com/littlevgl/lvgl/master/lv_misc/lv_fonts/lv_font_dejavu_40.c

Your problem can probably be solved by adjusting the `uint8_t *` variables in your code to be `const uint8_t *`.

embeddedt, 2018-10-04:
LVGL font bitmaps are stored as `const`. I think the problem may be that your pointers do not reflect this.

ywkelvin, 2018-10-07:
It works. Thanks for your help !

closed 2018-10-07 by ywkelvin