lvgl/lvgl #444: lv_theme_material_init() not found when compiling

PalantirWielder, 2018-10-05:
Using v5.1.1 of lvgl, every other theme init function was working fine, but lv_theme_material_init() wasn't found.

Traced the problem back to lv_themes/lv_theme.h:
```

/**********************
 *     POST INCLUDE
 *********************/
#include "lv_theme_templ.h"
#include "lv_theme_default.h"
#include "lv_theme_alien.h"
#include "lv_theme_night.h"
#include "lv_theme_zen.h"
#include "lv_theme_mono.h"
```

the addition of `#include "lv_theme_material.h"` resolved the issue

PalantirWielder, 2018-10-05:
v5.2 has the issue resolved
https://github.com/littlevgl/lvgl/blob/41695bf9ac758ae3276424e06308e1bb89c4c019/lv_themes/lv_theme.h#L302

closed 2018-10-05 by PalantirWielder

kisvegabor, 2018-10-06:
Thank you anyway! :)