lvgl/lvgl #446: Duplicatged LV_FONT_DEFAULT in configuration file

ErichStyger, 2018-10-06:
Hi,
In the latest lv_conf_templ.h the following define is present twice:
LV_FONT_DEFAULT
One of them should be removed.

Created pull request for it:
https://github.com/littlevgl/lvgl/pull/445

Thanks,
Erich

kisvegabor, 2018-10-06:
Hello,

I merged your PR. Thank you!

I'm planning to move the documentation of LittlevGL from the website to
GitHub wiki. Mainly it means copy pasting and reviews. Would you be
interested in it?
I will open an issue on GitHub to get help in it.

Best regards,
Gabor

#445 <https://github.com/littlevgl/lvgl/pull/445>

ErichStyger, 2018-10-06:
Having the documentation on Git is a good idea. For sure I help if I can. My time will be limited, but if I spot something or if I have anything to add, I'm happy to contribute.

closed 2018-10-07 by kisvegabor