lvgl/lvgl #447: Move the documentation to GitHub Wiki

kisvegabor, 2018-10-07:
Hi,

I decided to move the documentation from https://littlevgl.com to here as GitHub Wiki Pages. It will enable everybody to edit and extend the documentation if something is missing or not explained well enough.

At first, moving the documentation means mainly copy-pasting the existing texts from https://littlevgl.com to here. 
To get started I created the skeleton of the Wiki documentation (https://github.com/littlevgl/lvgl/wiki) and added references to current documentation.

The contribution would be very welcome! If you have some spare time just pick a Wiki page and copy-paste the existing documentation and format it using [Markdown](https://guides.github.com/features/mastering-markdown/). 

Please drop a line here is you would like help in it!

Thank you in advance!

embeddedt, 2018-10-07:
@kisvegabor I can help with this, but I seem to be unable to edit the wiki at this point. Possibly it's restricted to collaborators only?

Zaltora, 2018-10-07:
Do you know [mdBook](https://github.com/rust-lang-nursery/mdBook) ? Rust documentation is generated with this. It transform markdown files based documentation into modern web page.

kisvegabor, 2018-10-08:
@embeddedt 
True, it was restricted to collaborators only. I enabled it.

@Zaltora  
I didn't know mdbook but it seems cool! Although I'd like to keep the documentation in one place and not having a website too. As some people asked for an offline (pdf) documentation do you known a good markdown->pdf generator?

kisvegabor, 2018-10-08:
True, an HTML documentation can be offline as well. Sound well!

embeddedt, 2018-10-08:
@kisvegabor Could you please do one page as an example so we know the formatting to use?

seyyah, 2018-10-08:
> As some people asked for an offline (pdf) documentation do you known a good markdown->pdf generator?

There are a lot of tools for this purpose [1],

- github-wikito-converter [2]
- https://github.com/jobisoft/wikidoc [3]
- pandoc [4]

## :clipboard: References
1. https://medium.com/@yakivmospan/did-you-ever-try-to-convert-github-wiki-in-pdf-c3078538bf12
2. https://github.com/yakivmospan/github-wikito-converter
3. https://github.com/jobisoft/wikidoc
4. http://pandoc.org/index.html

kisvegabor, 2018-10-09:
@embeddedt 
A created two pages as examples:
https://github.com/littlevgl/lvgl/wiki/Styles
https://github.com/littlevgl/lvgl/wiki/Label

To add images I cloned the wiki repo and copied the images from the website to the `img` folder and used them like this:
`
[image]` 

@seyyah 
Thank you for this nice collection of tools!

C47D, 2018-10-09:
Do you think is possible to add documentation in non English languages?

embeddedt, 2018-10-09:
@kisvegabor Thanks for the two examples! They will be very helpful!

How would I go about adding an image to the wiki repository? I looked it up and it seems that Github doesn't support pull requests on wikis...

https://stackoverflow.com/a/11481887

kisvegabor, 2018-10-09:
@C47D 
I found this: https://webapps.stackexchange.com/questions/30567/internationalization-of-the-github-wiki-or-github-pages
It seems it's not simple to manage but possible

@embeddedt 
Uh, really. It's not so convenient. I think the most simple way would be if I added the images myself and then you can use them.

Zaltora, 2018-10-09:
A solution used by esp-idf project:
[readthedocs](https://readthedocs.com/)
the esp-idf documentation:
[esp-idf documentation](https://docs.espressif.com/projects/esp-idf/en/latest/)
the sources:
[sources docs](https://github.com/espressif/esp-idf/tree/master/docs)

It is support version, localization and others things. the file format is differents, it is `.rst` (reStructuredText)

C47D, 2018-10-09:
@kisvegabor Maybe doing like [this](https://github.com/Obijuan/open-fpga-verilog-tutorial/wiki) other wiki, all the topics have a (EN) link on the side which leads to the the page translated to ENglish.

@Zaltora does the esp-idf project uses [Sphinx](http://www.sphinx-doc.org/en/master/) for its documentation?

Zaltora, 2018-10-09:
@C47D  I guess yes. Main page doc from esp-idf: `Built with Sphinx using a theme provided by Read the Docs. `

embeddedt, 2018-10-11:
@kisvegabor I've just created the Image page on the wiki.

https://github.com/littlevgl/lvgl/wiki/Image

I have added everything except the image for the example. Could you please make sure it looks all right?

kisvegabor, 2018-10-11:
@embeddedt  Awesome! I can add the images tomorrow.

@C47D For multi-language support: It's simple and it can work! Let's do this way.

@Zaltora A I see `readthedoc` doesn't have a free plan. Is it correct?

C47D, 2018-10-12:
@kisvegabor, great, i will try to translate the English pages into Spanish when they are finished.

In regards of readthedocs, as far as I know they host open source projects for free.

kisvegabor, 2018-10-12:
@C47D It would be great!

kisvegabor, 2018-10-12:
I pushed the images and updated the Image's documentation with the example image.

embeddedt, 2018-10-12:
@kisvegabor Thanks! I'll do Line, Arc, and Container next.

embeddedt, 2018-10-12:
@kisvegabor I realized that about 5 minutes after writing the message. :(

So I converted Line, Container, Page, Window, and Tabview. They are still missing images, of course.

kisvegabor, 2018-10-12:
@embeddedt Great work! 
Were the images not working this way: `
[image]`?

embeddedt, 2018-10-12:
@kisvegabor Oops. I thought the images hadn't been added yet. Fixing that now!

kisvegabor, 2018-10-12:
Awesome, thank you very much! 
I will add some of the new of the new objects next week.

embeddedt, 2018-10-13:
@kisvegabor I've finished converting all the old object pages.

embeddedt, 2018-10-19:
@kisvegabor All of the documentation pages have now been filled in. We still need to add the new features.

kisvegabor, 2018-10-27:
@embeddedt Sorry for the late answer.
I added the 4 new object types (arc, preload, calendar, image button)

C47D, 2018-10-27:
@kisvegabor Should i start adding pages translated to Spanish?

kisvegabor, 2018-10-28:
@C47D It's perfect! We can do it this way!

@seyyah I saw that you set alphabetic order in the sidebar. It's a good idea for the object types but I think for the introductions (porting, objects etc)  we should keep the order as the user should learn the library. Do you agree?

embeddedt, 2018-10-28:
How will we keep the translated pages in sync with the English version?

C47D, 2018-10-28:
@embeddedt Good question, in my case i was planning to do it manually updating the pages on the weekends, using google translate is a good alternative (i think most of us non-native English speakers do that).

kisvegabor, 2018-10-28:
@embeddedt Good question indeed. As nobody can guarantee that the documentation will be up to date 
 in all languages all the time I can imagine to add an lvgl version number on the top of every page. This way people can see that the documentation is outdated. What do you think?

seyyah, 2018-10-29:
> @seyyah I saw that you set alphabetic order in the sidebar. It's a good idea for the object types

I think also like you @kisvegabor 

> but I think for the introductions (porting, objects etc) we should keep the order as the user should learn the library. Do you agree?

I agree and change it

[image]

What do you think about [Coding Style Guide](https://github.com/littlevgl/lvgl/wiki/Coding-Style-Guide)

seyyah, 2018-10-29:
> @kisvegabor Should i start adding pages translated to Spanish?
> 
> EDIT:
> I did added the translation of the Home page as a test, here is how i did so:
> 
> 1. Edit the wiki sidebar.
> 2. Added ([[ES|Home_es]]) at the end of the Home line, Home_es is the name of the Spanish translation of the Home page,

I think that we use different directory for each language. For example, [ArchiSteamFarm](https://github.com/JustArchiNET/ArchiSteamFarm/wiki)

English,
[image]

Russia,
[image]

Directory tree is
[image]

Locale directory is
[image]

For example, tr-TR directory is
[image]

We use footer for translation links.
[image]

Note: I added the test contentn and commented it.

kisvegabor, 2018-10-30:
@seyyah  

> What do you think about Coding Style Guide?

I like it! And the new order is great as well!

> I think that we use different directory for each language. For example, ArchiSteamFarm

I like it as well. If we would have a lot of languages the sidebar could look ugly with a lot of parentheses. (like `Home (ES)(CN)(FR)(DE)(HU)`). I didn't know that you can have multiple `_Sidebar.md` in multiple directories. This way the sidebar can be translated as well.  The flags look great too!

C47D, 2018-10-30:
> I like it as well. If we would have a lot of languages the sidebar could look ugly with a lot of parentheses. (like Home (ES)(CN)(FR)(DE)(HU)). I didn't know that you can have multiple _Sidebar.md in multiple directories. This way the sidebar can be translated as well. The flags look great too!

Agree, the flags looks great.

kisvegabor, 2018-10-31:
@C47D Great! Then can you try to add the Spanish Home page this way?

C47D, 2018-11-01:
@kisvegabor I will later tonight, but i'm not sure where to locate the locale directory, should i create a clone of the Wiki and add it that way or can you help me with that?

I'm located in Mexico, so i don't know if the Spanish language directory should be es_MX or es_LAT (LAT for Latinoamerica).

kisvegabor, 2018-11-01:
@C47D  Yes, you can clone the wiki repo but I'm not sure if you can push it back. For simplicity, I tried the new directory structure and it seems working. I added the `locale/es_MX` directory,` Home_es_MX.md` and a local `_Sidebar`.

If you have no permission to push to the wiki repo I can create the pages with `es_MX` suffix.

> I'm located in Mexico, so i don't know if the Spanish language directory should be es_MX or es_LAT (LAT for Latinoamerica).

I've just chosen "es_MX" while testing but without any reason. Do you know which one is more common? And can you tell how different `es_MX` is from `es_ES`?

seyyah, 2018-11-01:
[image]

> - [ISO 639-1](http://en.wikipedia.org/wiki/ISO_639-1) – covers the registration of two letter language codes. The complete list of all registered ISO 639-1 language codes can be found [here](http://en.wikipedia.org/wiki/List_of_ISO_639-1_codes).

> - language / region separators that we accept are dash and underscore ( – _ )

> - [ISO 3166-1 alpha-2 codes](http://en.wikipedia.org/wiki/ISO_3166-1_alpha-2) – defines two letter country codes that represent countries, dependent territories, and special areas of geographical interest.

[image]

## :clipboard: Reference
1. https://lingohub.com/blog/2014/03/lingohub-support-language-codes-localization/
2. https://wpastra.com/docs/complete-list-wordpress-locale-codes/

seyyah, 2018-11-01:
We want to add Turkish i18n. But we have not enough permission. So we create a gist.

Gist: https://gist.github.com/seyyah/f2a0c5a6bdecda5897e931e5c6e75e33

## :clipboard: Reference
1. https://stackoverflow.com/questions/10642928/how-to-pull-request-a-wiki-page-on-github

C47D, 2018-11-01:
> @C47D Yes, you can clone the wiki repo but I'm not sure if you can push it back. For simplicity, I tried the new directory structure and it seems working. I added the `locale/es_MX` directory,` [...]

I had never been in Spain (ES) but in Latinoamerica it's almost the same lang, but with the help of @seyyah answer i suggest to use `es` alone, this is the mandatory language code field in ISO 639-1.

kisvegabor, 2018-11-02:
> @seyyah answer i suggest to use es alone, this is the mandatory language code field in ISO 639-1.

Based on this what about using only the language code for other languages too. It's very improbable that the documentation will be translated into several regions of a language. Do you agree?

@seyyah Thank you for this useful overview of this topic! I've just added you gist as `tr`.

C47D, 2018-11-02:
> > @seyyah answer i suggest to use es alone, this is the mandatory language code field in ISO 639-1.
> 
> Based on this what about using only the language code for other languages too. It's very improbable that the documentation will be translated into several regions of a language. Do you agree?

I do agree @kisvegabor, i will also upload files in a gist, i had already started to translate some.

kisvegabor, 2018-11-04:
@C47D  Awesome!

As we discussed it I added the `Written for v5.1` line to each English page to show which is the related LittelvGL version.

embeddedt, 2018-11-04:
@kisvegabor In order to handle the case where a page is edited between release versions I propose that we should add a revision number to each page. That way it is immediately clear if the translation differs from the English version.

As an example, I just added a note about X/Y coordinates to the Porting page, but this will not be reflected in translations and the difference will not be obvious to users of those translations.

kisvegabor, 2018-11-06:
@embeddedt It really could be a problem. I like the proposed revision number. So what about something like:
_Written for v5.1 (rev. 5)_

embeddedt, 2018-11-06:
@kisvegabor Exactly what I was thinking!

kisvegabor, 2018-11-06:
The only issue could be that everybody has to increment the revision number when he edits the documentation. It should be clearly stated somewhere.

embeddedt, 2018-11-06:
Maybe there is a way to block edits till rules are read and understood.

seyyah, 2018-11-07:
We suggest `_Footer`, `_Sidebar` content as below,

```markdown
| [<img ...>](../wiki/Home) | [<img ...>](../wiki/Home_es)| [<img ...>](../wiki/Home_tr) |
|-----|-----|-----|
```
**Note**: `../wiki/Home_es` link shows that `https://github.com/littlevgl/lvgl/wiki/Home_es.

In _Sidebar.md

```markdown
* [[Home]]
* ...
* [[Object types]]
  * [Arc (lv_arc)](../wiki/Arc)
  * ...
```

In locale/_Sidebar.md,

```markdown
* [Anasayfa](Home_tr)
* ...
* [Nesneler](../wiki/Object-types_tr)
  * [Arc (lv_arc)](../wiki/Arc_tr)
```

seyyah, 2018-11-07:
In addition to we suggest to add version control

- `[Arc (lv_arc)](../wiki/Arc)`: the latest version
- `[Arc (lv_arc)](../wiki/Arc_tr)`: the latest version for locale
- `[Arc (lv_arc) v5.1](../wiki/Arc-v5.1)`: specific version
- `[Arc (lv_arc) v5.1](../wiki/Arc-v5.1_tr)`: specific version for locale

In `Arc.md` file

```markdown
Current version: v5.2. Previous versions: [v5.1](./Arc-v5.1), [v5.0](./Arc-v5.0)
```

embeddedt, 2018-11-07:
@seyyah @kisvegabor This solution with revision numbers, footers, and sidebars is starting to seem very complex and convoluted. Maybe the GitHub wiki system isn't the best for what we're trying to accomplish.

embeddedt, 2018-11-09:
@kisvegabor 

> the GitHub Wiki can work except the revision numbers.

Agreed.

> What about a line in the footer like this? `If you don't find something on the translated page, have a look at the original English version.` It's not a real solution but can work.

Yes, seems a little hacky, but it will work fine. :+1: 

> The user won't know if a feature exists in v5.0 or it's only in v5.1. But is it really an issue? 

I don't think it's an issue. If the user has a need to stay on an older version (#483) that should probably be addressed because there shouldn't be such groundbreaking changes. However, that being said, 6.0 is going to break things and I think that needs to be made clear somewhere. Maybe a statement on the home page stating that backwards compatibility is only guaranteed across minor versions?

seyyah, 2018-11-09:
> @seyyah I think the `v5.0`, `v5.1` links are not required because the library should be backward compatible among the minor versions. So in an ideal case, the `v5.1` documentation of the Arc object should be `v5.0` documentation + `new features`. However, the user won't know if a feature exists in `v5.0` or it's only in `v5.1`. But is it really an issue? Because the user should update the library to the new minor version.

Ok :+1:

embeddedt, 2018-11-09:
@kisvegabor Perfect!

seyyah, 2018-11-12:
Can you add `_Sidebar.md` for Turkish locale?

Gist: https://gist.github.com/seyyah/cf87bf4e68048d5e57645464b515596f

**Note**: The user without collobarator cannot edit locale/_Sidebar.md.

seyyah, 2018-11-12:
Thanks a lot

seyyah, 2018-11-12:
We have just added some locale files but the wiki make incorrect form our files

[image]

Our files are Arc_tr.md and Bar_tr.md in the root directory. When i opened one of these files, wiki change immediately _Sidebar to English locale. But these files should actually be in the locale/tr folder.

If you add a collobaration, i change it.

embeddedt, 2018-11-12:
@seyyah I can't add you as a collaborator without permission from @kisvegabor. But I can move those files to the `locale/tr` directory for you.

seyyah, 2018-11-12:
No problem. We take your time, sorry

[42 more replies]

closed 2018-12-02 by kisvegabor