lvgl/lvgl #449: set date by calendar

phoenix2000, 2018-10-09:
Hi
I want to set the date of RTC by calendar object. It seems that it doesn't have any lv_calendar_set_action function which then we can get the touched day number and decide to set it as the day of the date.
so now what should we do? Is there any other way to do this?

kisvegabor, 2018-10-09:
Hi,

`lv_calendar` is an "output object" right now and not supports touch and actions. However, it would be useful to add support for it! The contribution is very welcome in it. I kindly help if you want to add this feature.

phoenix2000, 2018-10-11:
OK. I'm ready and working on it.

kisvegabor, 2018-10-12:
Awesome! :)

These lines refer to the "buttons" on the header to change to the next/prev months. 

To select a new day you should do something similar: search the pressed day, and change to it on release.
Note that the position of the days depends on the styles of the calendar. Maybe you should check `draw_days` function to see how the coordinates are calculated.

phoenix2000, 2018-10-13:
we call the lv_obj_invalidate function to redraw the object. 
As much as I understand from codes, going to the next or previous month will happen on LV_SIGNAL_RELEASED. 
So then why do we call  lv_obj_invalidate  in other places? for example in 
```c
        if(lv_area_is_point_on(&header_area, &p)) {
            if(p.x < header_area.x1 + lv_area_get_width(&header_area) / 2) {
                if(ext->btn_pressing != -1) lv_obj_invalidate(calendar);
                ext->btn_pressing = -1;
            } else {
                if(ext->btn_pressing != 1) lv_obj_invalidate(calendar);
                ext->btn_pressing = 1;
            }
        }
```
why it shouldn't be like this:
```c
        if(lv_area_is_point_on(&header_area, &p)) {
            if(p.x < header_area.x1 + lv_area_get_width(&header_area) / 2) {
                ext->btn_pressing = -1;
            } else {
                ext->btn_pressing = 1;
            }
        }
```

kisvegabor, 2018-10-13:
When the next/prev buttons are pressed the invalidation is required to redraw with the pressed state.
Due to the invalidation, `draw_header` will be called and will draw the buttons with their current style (pressed/released).

phoenix2000, 2018-10-13:
you are right but here I don't see any changes between pressed and released state. I mean there is no pressed state of header button in default. do you mean we can set it in some other places?

kisvegabor, 2018-10-13:
The `LV_CALENDAR_STYLE_HEADER` and `LV_CALENDAR_STYLE_HEADER_PR` styles set the style of the arrows. So:
- `LV_CALENDAR_STYLE_HEADER` gives the released arrow color and the header style
- `LV_CALENDAR_STYLE_HEADER_PR` gives the style of the arrow (only `style.text` attributes are used)

By default, the pressed arrow should be a little bit grayer.

kisvegabor, 2018-11-02:
As the PR is merged I close this issue! #460

Thank you for the contribution!

closed 2018-11-02 by kisvegabor