lvgl/lvgl #450: lv_list/lv_slider has compilation errors when USE_LV_GROUP is disabled

embeddedt, 2018-10-09:
Hi,

When compiling LVGL `master` with `LV_USE_GROUP` disabled, there are compilation errors due to missing types and variables that are only enabled when `LV_USE_GROUP` is.

It looks like these might be fixable by wrapping them in preprocessor statements. I have submitted a PR that tries to do that, but I am still fairly new to LVGL so I may have done something incorrectly or disabled too much code.

From what I could tell, these may have been introduced by the encoder changes in #422 .

I have tested and the list still seems to work with my changes.

```
lvgl/lv_objx/lv_list.c: In function 'lv_list_signal':
lvgl/lv_objx/lv_list.c:646:13: error: unknown type name 'lv_group_t'; did you mean 'lv_disp_t'?
             lv_group_t * g = lv_obj_get_group(list);
             ^~~~~~~~~~
             lv_disp_t
lvgl/lv_objx/lv_list.c:656:16: error: 'last_clicked_btn' undeclared (first use in this function)
             if(last_clicked_btn) {
                ^~~~~~~~~~~~~~~~
lvgl/lv_objx/lv_list.c:656:16: note: each undeclared identifier is reported only once for each function it appears in
lvgl/lv_objx/lv_list.c:667:14: error: 'lv_list_ext_t {aka struct <anonymous>}' has no member named 'selected_btn'; did you mean 'styles_btn'?
         ext->selected_btn = NULL;
              ^~~~~~~~~~~~
              styles_btn
lvgl/lv_objx/lv_list.c:675:21: error: 'lv_list_ext_t {aka struct <anonymous>}' has no member named 'selected_btn'; did you mean 'styles_btn'?
             if(ext->selected_btn) {
                     ^~~~~~~~~~~~
                     styles_btn
lvgl/lv_objx/lv_list.c:676:71: error: 'lv_list_ext_t {aka struct <anonymous>}' has no member named 'selected_btn'; did you mean 'styles_btn'?
                 lv_obj_t * btn_prev = lv_list_get_next_btn(list, ext->selected_btn);
                                                                       ^~~~~~~~~~~~
                                                                       styles_btn
lvgl/lv_objx/lv_list.c:686:21: error: 'lv_list_ext_t {aka struct <anonymous>}' has no member named 'selected_btn'; did you mean 'styles_btn'?
             if(ext->selected_btn != NULL) {
                     ^~~~~~~~~~~~
                     styles_btn
lvgl/lv_objx/lv_list.c:687:71: error: 'lv_list_ext_t {aka struct <anonymous>}' has no member named 'selected_btn'; did you mean 'styles_btn'?
                 lv_obj_t * btn_next = lv_list_get_prev_btn(list, ext->selected_btn);
                                                                       ^~~~~~~~~~~~
                                                                       styles_btn
lvgl/lv_objx/lv_slider.c: In function 'lv_slider_signal':
lvgl/lv_objx/lv_slider.c:489:9: error: unknown type name 'lv_group_t'; did you mean 'lv_disp_t'?
         lv_group_t * g = lv_obj_get_group(slider);
         ^~~~~~~~~~
         lv_disp_t

```

kisvegabor, 2018-10-09:
Hi,

Thank you for pointing this out. I really forget to try what happens if some features are disabled.
I made some test and added some fixes. I checked you PR and I did something similar but I found other issues with other objects/features as well. I hope you don't mind if I don't use your PR but I use my version.

embeddedt, 2018-10-09:
@kisvegabor No problem! Glad to know it's fixed.

closed 2018-10-09 by embeddedt