lvgl/lvgl #461: png decoder cannot display properly

relliktei, 2018-10-16:
Now I follow this post https://littlevgl.com/blog/50/use-png-images-in-littlevgl to set my screen background:

my code is like this:

unsigned LoadPng(const char* filename, lv_img_dsc_t *png_dsc)
{
	uint32_t error;
	unsigned char * png_data;
	size_t png_data_size;
	unsigned ret = 1;
	error = lodepng_load_file(&png_data, &png_data_size, filename);
	if(error == 0)
	{
		unsigned char * png_decoded;
		uint32_t png_width;
		uint32_t png_height;
		error = lodepng_decode32(&png_decoded, &png_width, &png_height, png_data, png_data_size);
		if(error == 0)
		{
			png_dsc->header.always_zero = 0;
			png_dsc->header.cf = LV_IMG_CF_TRUE_COLOR_ALPHA;
			png_dsc->header.w = png_width;
			png_dsc->header.h = png_height;
			png_dsc->data_size = png_width * png_height * 4;
			
			png_dsc->data = png_decoded;
			ret = 0;
		}
	}
	return ret;
}

create image object and display here:
----------------------------
#if WALLPAPER
	
	if(LoadPng("1:background.png", &png_dsc) == 0)
	{
		imgBackground = lv_img_create(scr, NULL);
		lv_img_set_src(imgBackground, &png_dsc);
		lv_obj_set_size(imgBackground, LV_HOR_RES, LV_VER_RES);
		lv_obj_set_pos(imgBackground, 0, 0);
		
	}
#endif
---------------------------------------------------------------------

I want to display this picture:
[image]
but the result is like this:
[image]

displayed image looks not good,  does anybody know where the problem is?

ivonask, 2018-10-16:
Hello,

I have a similar problem. I used the decoder mentioned in the previous comment, using the test image. After the initialization using png_decoder_init() from png_decoder.c, i did this simple try in my code:

`lv_obj_t* try_image = lv_img_create(scr_, NULL);
lv_img_set_src(try_image, "lib/display/littlevGL/png_decoder/png_decoder_test.png");`

In the decoder_info function I've set up the cf to LV_IMG_CF_TRUE_COLOR_ALPHA and got the following result:
[image]

I also tried LV_IMG_CF_TRUE_COLOR:
[image]

Any help would be appreciated, thank you.

kisvegabor, 2018-10-16:
What is your color depth? `LV_COLOR_DEPTH` in `lv_conf.h`.
By default, the png decoder in the post decompresses the images to 32 bit ARGB888 format. Which is not good if you have 16 bit RGB565 color depth.

ivonask, 2018-10-16:
Yes, the color depth was set to 16. Adjusting it to 32 I get the following, which is better but still not good:
[image]

I also tried using a different lodepng function for decoding in decoder_open()
`error = lodepng_decode_memory(&png_decoded, &png_width, &png_height, png_data, png_data_size, LCT_RGBA, 16);`
with this as a result:
[image]

Could you please explain how to use ARGB888 format, or what to do to make 16 bit RGB565 work? Thank you.

kisvegabor, 2018-10-17:
Hi,
I've just updated the png_decoder. Please download it again.
Now it converts the output image if you don't have 32 bit color depth.

kisvegabor, 2018-10-17:
https://littlevgl.com/blog/png/png_decoder.zip

ivonask, 2018-10-18:
> Hi,
> I've just updated the png_decoder. Please download it again.
> Now it converts the output image if you don't have 32 bit color depth.

Thank you! I would just like to add that since the type of my image was a file, i needed to add `convert_color_depth(png_decoded,  png_width * png_height);` before `return png_decoded;` in the `if (src_type == LV_IMG_SRC_FILE)` block of the decoder_open function.

kisvegabor, 2018-10-19:
I added updated the decoder again by adding the converter to the file decoding as you pointed out. 

This color order worked for me on PC but glad to hear you made it work.

It seems this improvement solved the issue so I close this issue. If you have remarks feel free to comment here.

closed 2018-10-19 by kisvegabor