lvgl/lvgl #462: Image No data

Zaltora, 2018-10-16:
Since i update lvgl 5.2, i got a problem. I can't display data from my file.

I converted an image with the tool to binary data (for monochrome display).
The conversion setting:  Indexed 2 colors  / Binary output.
Adding file system:
```c
    lv_fs_drv_t pcfs_drv;                         /*A driver descriptor*/
    memset(&pcfs_drv, 0, sizeof(lv_fs_drv_t));    /*Initialization*/
    pcfs_drv.file_size = sizeof(pc_file_t);       /*Set up fields...*/
    pcfs_drv.letter = 'P';
    pcfs_drv.open = _xEspSpiffsOpen;
    pcfs_drv.close = _xEspSpiffsClose;
    pcfs_drv.read = _xEspSpiffsRead;
    pcfs_drv.seek = _xEspSpiffsSeek;
    pcfs_drv.tell = _xEspSpiffsTell;
    lv_fs_add_drv(&pcfs_drv);
```
Load image:
```c
    pxPrvArgs->apxObjects[0] = lv_img_create(lv_scr_act(), NULL);
    lv_img_set_src( pxPrvArgs->apxObjects[0], "P:/logoCS.bin");
```

When i list file in my filesystem with my command, i see the file and the correct size.  What i miss with 5.2 changes ?

kisvegabor, 2018-10-16:
Hi,

>  I can't display data from my file.
What does "can't display" means exactly?

What if you convert the image to true color instead of indexed?

Zaltora, 2018-10-16:
New conversion setting: true  color / Binary RGB332 output.
My monochrome screen is fully light.

kisvegabor, 2018-10-16:
Please send the image you want to convert.

Zaltora, 2018-10-16:
This one : [Logo_C_S_HautNB.xbm.zip](https://github.com/littlevgl/lvgl/files/2483420/Logo_C_S_HautNB.xbm.zip)

kisvegabor, 2018-10-16:
Only png, jpg and bmp images are supported. I converted your image to png and converted it to C array and looks good now. the Binary version should work too.

Here is the png image:
[image]

Zaltora, 2018-10-16:
I confirm, it is work! thank you.

closed 2018-10-16 by Zaltora