lvgl/lvgl #463: touch-aware line meter?

mharizanov, 2018-10-16:
Hello,
I would like to be able to use the touch functionality to set the line meter value i.e. touch a certain point of the arc of the line meter and this way set its value. 
What are my best options?
Thanks :)

kisvegabor, 2018-10-17:
Hi,

Unfortunately, I can't say an off-the-shelf solution. To achieve this the Line meter object should be improved. If you are interested in this feature I kindly to implement it.

mharizanov, 2018-10-17:
Thanks Gabor,
I suspected it will be a complex thing. For now I'll settle for a roller instead, and if I change my mind, I'll contact you:

[[image]]
(https://www.youtube.com/watch?v=4M9Ncon5PV8)

closed 2018-10-17 by mharizanov

kisvegabor, 2018-10-17:
Looks great!

I can imagine an animation on setting the values. Maybe it's worth a try:
```c
    lv_obj_t * lm = lv_lmeter_create(lv_scr_act(), NULL);

    lv_anim_t a;
    a.var = lm;
    a.start = 0;
    a.end = 50;
    a.fp = (lv_anim_fp_t)lv_lmeter_set_value;
    a.path = lv_anim_path_linear;
    a.end_cb = NULL;
    a.act_time = 0;
    a.time = 200;
    a.playback = 0;
    a.playback_pause = 0;
    a.repeat = 0;
    a.repeat_pause = 0;
    lv_anim_create(&a);
```

mharizanov, 2018-10-17:
Ha! Implemented it and it looks lovely.. thanks for the tip! My only remaining issue is that sometimes the roller stops in a mid-position, not snapped to a selection. Is that a known issue?

kisvegabor, 2018-10-17:
"mid-position" means stops between two options? If yes, it's not known, I never experienced it.

mharizanov, 2018-10-17:
Yes, exactly. I fiddled with **LV_INDEV_DRAG_LIMIT**, increased it to 20 and it seems fine now.. probably something specific to my touch implementation.

köszönöm

kisvegabor, 2018-10-18:
Great! 

Szívesen :)