lvgl/lvgl #464: Label text justification

DeonMarais64, 2018-10-17:
Hi Gabor,

Please consider if you will the following text justification layouts,
The font used would be one of the defaults, in this case lv_font_dejavu_20

Layout 1,
```
Display Module
    Good packets per second : 120.0
    Request count           :   15426391
    Good packets            :   15297303,  99.16%
    Bad packets             :          0,   0.00%
    Not addressed packets   :          0,   0.00%
    No response             :     129088,   0.84%
    Run count               :   30561677
Field Module
    Good packets per second : 113.8
    Request count           :   14630007
    Good packets            :   14515661,  99.22%
    Bad packets             :         31,   0.00%
    Not addressed packets   :          0,   0.00%
    No response             :     114314,   0.78%
    Run count               :   29398852
```
As I understand things, to achieve this one would need to,

	1. Create a label with a ' ' (space) as a reference point (lbl_ref) at the desired justification point (to the left of where the collumns starting with "Good packets per second" (lbl_desc)) is,
	2. Call lv_obj_align(lbl_desc, lbl_ref, LV_ALIGN_OUT_RIGHT_MID)
	3. Create a label with a ' ' or ',' or as a reference point at the desired justification point (to the right of where the collumns starting with "15426391" (lbl_value)) is,
	4. Call lv_obj_align(lbl_value, lbl_ref, LV_ALIGN_OUT_LEFT_MID)

or something similar.

Layout 2,
```	
Display Module
    Good packets per second : 120.0
              Request count : 15426391
               Good packets : 15297303,    99.16%
                Bad packets : 0,            0.00%
      Not addressed packets : 0,            0.00%
                No response : 129088,       0.84%
                  Run count : 30561677
Field Module
    Good packets per second : 113.8
              Request count : 14630007
               Good packets : 14515661,    99.22%
                Bad packets : 31,           0.00%
      Not addressed packets : 0,            0.00%
                No response : 114314,       0.78%
                  Run count : 29398852
```

As I understand things, to achieve this one would need to,

	1. Create a label with a ':' as a reference point (lbl_ref) at the desired justification point (to the right of where the collumns starting with "Good packets per second" (lbl_desc)) is,
	2. Call lv_obj_align(lbl_desc, lbl_ref, LV_ALIGN_OUT_LEFT_MID)
	3. Call lv_obj_align(lbl_value, lbl_ref, LV_ALIGN_OUT_RIGHT_MID)
	
or something similar.

* Firstly is there a more elegant/efficient way to achieving this type of justification ?
* If not, would it be possible to add left, right and perhaps center justification options to label texts in the future ?
* I imagine this could be achieved either by adding other LV_LABEL_ALIGN options or adding a justification member to the text struct in lv_style_t.

Thanks in advance for your consideration of this matter and look forward to your response.

kisvegabor, 2018-10-17:
Hi,

For me, Layout 2 seems easier to implement in an elegant way. The key is you can use `\n` in the label's text to add a  line break
1. Create a label with `LV_LABEL_ALIGN_RIGHT` for the description with a text like this:
```  
"Good packets per second : \n
Request count : \n
Good packets : \n
..."
```
2. Crate a similar label for the values with `LV_LABEL_ALIGN_LEFT`
3. Crate a similar label for the percentages with `LV_LABEL_ALIGN_LEFT`
4. Set the psotion of the `desc` label with `lv_obj_set_pos(desc, 30, 20)` or align it to the   `Display Module` title wih `lv_obj_align(desc, title, LV_ALIGN_OUT_BOTTOM_LEFT, 30, 0)`
5. Align the `values` to the `desc`: `lv_obj_align(values, desc, LV_ALIGN_OUT_LEFT_TOP, 5, 0);`
6. Align the `pcts` to the `values`: `lv_obj_align(pcts, values, LV_ALIGN_OUT_LEFT_TOP, 5, 0);`

DeonMarais64, 2018-10-17:
Hi Gabor,
Thanks for your response,
The difficulty with that is one would need to deal with the layout order separatly from the lines.
I often have the need to display a number of eliments in a similar way to what I described above.
For example take the following,
<Description> <Value> <Units>
<Description> <.....Value....>
..
<Description> <Value> <Units>
<Description> <Value>
..
<Description> <Value> <Units>
Sometimes needing the Values to be right, left or center justified.
Where the description and units are static and value changes.
I have created descriptors and code that deals with them to set the layout up and then only update the values when they change. They way I do it is to have 3 separate labels.
Also sometimes different font sizes are involved as well.
As I see it it can be made to work the way that you describe, but it seems unwieldy to me. If there were a way to specify justification of lables, this would be made vastly simpler.

kisvegabor, 2018-10-17:
>  If there were a way to specify justification of lables, this would be made vastly simpler.

Do you mean to specifiy the justification line by line? Or even inside a line?
The current `lv_label` certainly can't do this. It can justify only the whole label with `lv_label_set_aling()`

Maybe what you would need is a tabular-like object (it doesn't exists right now) where each call can have different text and justification.

kisvegabor, 2018-10-17:
How do you mean: "justification of individual lables and elements inside a line would offer the greatest flexibility"? Let's say we have a  text: `"Speed1: 11 km/h"`. How to configure it? Tell the position of each word or "part"? For example: 
`"<100:right> Speed1 <100:left>: <150:right>11 <150:left>km/h"`
```
                        100               150
                   Speed |:             11 | km/h
```

DeonMarais64, 2018-10-17:
Yes, something of that nature that would be terrific.

kisvegabor, 2018-10-17:
As I see it rather looks like a tabular E.g `lv_tabular_set_cell_values(tabular1, 3, 7, "mA")`  to set the [3;7] cell to "mA". And `lv_tabular_set_cell_align(tabular, 3, 7, LV_LABEL_ALIGN_RIGHT )`.

DeonMarais64, 2018-10-17:
If there is support for justification of individual fields then it would pobably do the trick. 
Also it would be usefull to have an option to display with or without a table frame.

kisvegabor, 2018-10-17:
> Also it would be usefull to have an option to display with or without a table frame.

It's just the question of the styles :)

> So in a field could occupy a certain width, but each with its own justification option (left, right or centered).

Yes, exactly! If you are interested in implementing this object type, I gladly help in it.

DeonMarais64, 2018-10-17:
I am in the middle of a project wich needs it at the moment, once I am done I will gladly share it and we can refine it to make it as general purpose/flexible as possible.