lvgl/lvgl #466: Trig full screen refresh

ffzwadd, 2018-10-18:
This is not an issue, we need a new feature. I'm trying to port lvgl to Eink display. I have to write full screen data to the Eink twice. But I do not know the best way for that.
[WF0420T8OBF47E41eU76.pdf](https://github.com/littlevgl/lvgl/files/2490765/WF0420T8OBF47E41eU76.pdf)

embeddedt, 2018-10-18:
Could you please clarify what you mean by "write full screen data to the eInk twice?"

I hope I haven't misunderstood your question, but this may be able to offer some insight.

I'm not sure if LVGL has some API to allow you to know when it is finished all drawing. If it did, you might be able to implement your display driver this way:

1. Hold the display contents in a framebuffer. Whenever LVGL calls disp_map/disp_flush, simply draw into that framebuffer.
2. When LVGL is finished drawing, blit the framebuffer to your eInk display in whatever way your driver supports.

This might allow you to just blit one whole screen at a time to the eInk.

kisvegabor, 2018-10-19:
If you want to always refresh the whole screen you  can use a rounder function:
```c

void my_rounder(lv_area_t *a)
{
    a->x1 = 0;
    a->x2 = LV_HOR_RES - 1;

    a->y1 = 0;
    a->y2 = LV_VER_RES - 1;
}

...
    lv_refr_set_round_cb(my_rounder);
```

In this case, `LV_VDB_SIZE` should be `LV_HOR_RES * LV_VER_RES`. In your `disp_flush` function you will always get a screen sized area which you can send to display twice.

ffzwadd, 2018-10-29:
> If you want to always refresh the whole screen you can use a rounder function:
>
> ```c
> void my_rounder(lv_area_t *a)
> {
>     a->x1 = 0;
>     a->x2 = LV_HOR_RES - 1;
>
>     a->y1 = 0;
>     a->y2 = LV_VER_RES [...]

In my case, I don't have enough ram for doing that.

ffzwadd, 2018-10-29:
> Could you please clarify what you mean by "write full screen data to the eInk twice?"
>
> I hope I haven't misunderstood your question, but this may be able to offer some insight.
>
> I'm not sure if LVGL [...]

According to the spec. The physical screen resolution is 400*300. When we refresh a picture, we must follow the below flow:
1. Initial the eInk module controller;
2. Write command 10H (Data Start Transmission 1), in my setting it's white/black color data;
3. Write whole screen white/black data to the eInk module;
4. Write command 13H (Data Start Transmission 2), in my setting it's red color data;
5. Write whole screen red data to the eInk module;
6. Write command 12h to start refresh the screen;

kisvegabor, 2018-10-30:
I can imagine the following trick:
1. In `lv_refr.c` in `lv_inv_area` add this after `suc = lv_area_intersect(&com_area, area_p, &scr_area);`:
```c
    com_area.x1 = 0;
    com_area.y1 = 0;
    com_area.x2 = LV_HOR_RES - 1;
    com_area.y2 = LV_VER_RES - 1;
```
It will ensure that to always redraw the whole screen.

2. Declare a variable to know which is the current transmission: `bool red_transmission = false;`
3. Set up a refresh callback. It will be called every time when LittlevGL is ready with the refreshing. In the callback invalidate the screen again if `red_transmission == false` and toggle `red_transmission` .

```c  
lv_refr_set_monitor_cb(my_monitor_cb)

void my_monitor_cb(uint32_t time_ms, uint32_t px_num)
{
   if(red_transmission == false) {
       lv_obj_invalidate(lv_scr_act());
       red_transmission = true;
   } else {
     red_transmission = false;
   }
}
```

4. In your `disp_flush` function do the refresh according to `red_transmission`. 

This way LV_VDB_SIZE can be any value but you will get the rendered image in multiple chunks. 

I tryied in on PC and it worked.

ffzwadd, 2018-11-10:
> I can imagine the following trick:
> 
> 1. In `lv_refr.c` in `lv_inv_area` add this after `suc = lv_area_intersect(&com_area, area_p, &scr_area);`:
> 
> ```c
>     com_area.x1 = 0;
>     com_area.y1 = 0;
>     com_area.x2 = LV_HOR_RES - 1;
>     com_area.y2 = LV_VER_RES - 1;
> ```
> 
> It will ensure that to always redraw the whole screen.
> 
> 1. Declare a variable to know which is the current transmission: `bool red_transmission = false;`
> 2. Set up a refresh callback. It will be called every time when LittlevGL is ready with the refreshing. In the callback invalidate the screen again if `red_transmission == false` and toggle `red_transmission` .
> 
> ```c
> lv_refr_set_monitor_cb(my_monitor_cb)
> 
> void my_monitor_cb(uint32_t time_ms, uint32_t px_num)
> {
>    if(red_transmission == false) {
>        lv_obj_invalidate(lv_scr_act());
>        red_transmission = true;
>    } else {
>      red_transmission = false;
>    }
> }
> ```
> 
> 1. In your `disp_flush` function do the refresh according to `red_transmission`.
> 
> This way LV_VDB_SIZE can be any value but you will get the rendered image in multiple chunks.
> 
> I tried in on PC and it worked.

This works for me, Thanks.

closed 2018-11-10 by ffzwadd

kisvegabor, 2018-11-10:
Awesome!

byc013, 2020-03-04:
Hi,

I cannot find the declaration of lv_refr_set_monitor_cb() anymore. Is there a substitute for the same purpose in the latest API?

embeddedt, 2020-03-04:
Yes. Did you look at the [documentation](https://docs.littlevgl.com/en/html/porting/display.html#display-driver)?

> **monitor_cb** a callback function tells how many pixels were refreshed in how much time.

```c
disp_drv.monitor_cb = my_monitor_cb;
void my_monitor_cb(lv_disp_drv_t * disp_drv, uint32_t time, uint32_t px)
{

}
```