lvgl/lvgl #468: How can set Label position and size with parent of Page?

fallstool, 2018-10-19:
Hi
When I create label with the page container, can not set the correct postion and size. Is there some wrong with my code? thank you.
Here is the test code? 
    lv_obj_t * page1 = lv_page_create(lv_scr_act(), NULL);
    lv_page_set_scrl_layout(page1, LV_LAYOUT_OFF);
    lv_obj_set_size(page1, 320, 240);

    lv_obj_t * label = lv_label_create(page1, NULL);
    lv_obj_set_pos(label, 50, 50);
    lv_obj_set_size(label, 150, 50);
    lv_label_set_text(label, "First line of a text\n"
                      "Second line of a text\n"
                      "Third line of a text\n");

BrianPugh, 2018-10-19:
Do the notes at the bottom of the following link help?

https://littlevgl.com/object-types/page-lv_page

It may have to do with enabling/disabling scrolling with autofit.

fallstool, 2018-10-22:
Hi @BrianPugh 
I test with     lv_page_set_scrl_fit(page1, false, false);
It is not work for me .

kisvegabor, 2018-10-27:
Try this:
```c
    lv_obj_t * page1 = lv_page_create(lv_scr_act(), NULL);
    lv_obj_set_size(page1, 320, 240);
    lv_page_set_scrl_fit(page1, false, false);
    lv_page_set_scrl_height(page1, 200);

    lv_obj_t * label = lv_label_create(page1, NULL);
    lv_obj_set_pos(label, 30, 60);
    lv_label_set_text(label, "First line of a text\n"
    "Second line of a text\n"
    "Third line of a text\n");
```

[image]

fallstool, 2018-10-29:
@kisvegabor thank you.

kisvegabor, 2018-10-29:
Your welcome!
If it worked please close the issue.

closed 2018-10-30 by fallstool