lvgl/lvgl #469: How to make backspace button on a physical keyboard?

TMarkivULCRobotics, 2018-10-19:
I have a keyboard that is used for navigation through the objects, and for entering text into text fields. There is a definitions for navigation buttons in the lv_group.h file but how to implement backspace button  and delete button for text fields?

TMarkivULCRobotics, 2018-10-19:
And issue number two with the text area:
    When the a text is selected through the navigation buttons, then all touches on the text area add spaces to the text area. How to avoid that?

kisvegabor, 2018-10-19:
So do you need Delete and Backspace buttons too?

> all touches on the screen add characters to the text area

How do you mean that?

TMarkivULCRobotics, 2018-10-19:
Yes I need Delete and Backspace buttons.

I mean that whenever I touch the screen it adds some characters to the text area that is focused. To focus the text area I use navigation buttons.

kisvegabor, 2018-10-19:
> Yes I need Delete and Backspace buttons.

I've just added `LV_GROUP_KEY_BACKSPACE` to `dev-5.3` branch. Now there are separate keys for Delete and Backspace and the Text area can handle them.

> I  mean that whenever I touch the screen it adds some characters to the text area that is focused. To focus the text area I use navigation buttons.

CAn you send a video about it?

TMarkivULCRobotics, 2018-10-19:
Try this link to see the video:
https://drive.google.com/open?id=1fx2TPCrfDjncNYtctiXY5RtSWvDLdqtu

kisvegabor, 2018-10-20:
I don't experience this issue.
PLease send you touchpad driver too.

TMarkivULCRobotics, 2018-10-23:
Have you had a chance to reproduce the issue?

kisvegabor, 2018-10-27:
When the touchpad is released you should use the last pressed coordinate. Like here: https://github.com/littlevgl/lv_drivers/blob/master/indev/mouse.c#L55
However, it can't cause the issue.

It looks like the touchpad works as a keypad. Does the touchpad work in other cases?

 When you register the input devices have you set there type? 
`indev_drv.type = LV_INDEV_TYPE_POINTER;`

TMarkivULCRobotics, 2018-10-29:
1 a) No I am not using a touch pad as a keyboard.
   b) What do you mean touchpad work in other cases?
2 Yes I setup the drv type as you stated.

Below is tha keypad and touchpad init functions:
```
uint32_t keypad_init(void)
{
    uint32_t status = tca8418e_init();
    if (STS_SUCCESS == status)
    {
        lv_indev_drv_t  kp_drv;
        lv_indev_drv_init(&kp_drv);
        kp_drv.type = LV_INDEV_TYPE_KEYPAD;
        kp_drv.read = keypad_read;
        kp_indev = lv_indev_drv_register(&kp_drv);
    }

    return status;
}
```
```
uint32_t touchpad_init(void)
{
    uint32_t status = tsc2007_init();

    if (status == STS_SUCCESS)
    {
        /* Initialize library. Set touchpad function pointers. */
        lv_indev_drv_t indev_drv;
        lv_indev_drv_init(&indev_drv);
        indev_drv.read = touchpad_read;
        indev_drv.type = LV_INDEV_TYPE_POINTER;
        lv_indev_drv_register(&indev_drv);
    }

    return status;
}
```

kisvegabor, 2018-10-31:
> b) What do you mean touchpad work in other cases?

I mean can you use the touchpad properly to click buttons, swipe roller etc?

TMarkivULCRobotics, 2018-10-31:
Yes, in other cases touchpad works fine.

kisvegabor, 2018-11-02:
I see.
Your codes seems fine I have no idea what causes it.

Can you create a simple GUI to demonstrate the issue? It'd be the best if you could reproduce the issue in the PC simulator too.

TMarkivULCRobotics, 2018-11-02:
Ok I'll provide a code example, later today or tomorrow.

TMarkivULCRobotics, 2018-11-02:
Did you create a group object for the text area object when you were trying to reproduce the issue?

kisvegabor, 2018-11-04:
I used the `lv_test_group_1()` fro from `lv_examples` repo

TMarkivULCRobotics, 2018-11-08:
Hello Kris, Sorry for such delayed response. Here is the sample code that allows me to reproduce the issue:

```
lv_obj_t *text;

lv_res_t btn_callback(lv_obj_t *btn)
{
    lv_obj_t * label = lv_obj_get_free_ptr(btn);

    lv_label_set_text(label, lv_ta_get_text(text));

    return LV_RES_OK;
}

/* This may be main function. */
{
    lv_init();
    tft_init();
    touchpad_init();
    keypad_init();

    /* Create a text area. */
    text = lv_ta_create(lv_scr_act(), NULL);
    lv_ta_set_one_line(text, false);
    lv_ta_set_cursor_type(text, LV_CURSOR_LINE);
    lv_obj_set_width(text, 100);
    lv_obj_set_height(text,100);
    
    /* Create a label. */
    lv_obj_t *label = lv_label_create(lv_scr_act(), NULL);
    lv_label_set_text(label, "Initial text");
    
    /* Create a button. */
    lv_obj_t * btn = lv_btn_create(lv_scr_act(), NULL);
    lv_btn_set_toggle(btn, false);
    lv_btn_set_action(btn,
                      LV_BTN_ACTION_CLICK,
                      btn_callback);
    lv_obj_set_free_ptr(btn, label);

    /* Create a group so the text area will be available for editing. */
    lv_group_t * group = lv_group_create();
    lv_group_add_obj(group, text);

    /* You know what is it for ☺ */
    lv_obj_align(text, NULL, LV_ALIGN_CENTER, 0,0);
    lv_obj_align(label,text, LV_ALIGN_OUT_TOP_MID, 0,0);
    lv_obj_align(btn,  text, LV_ALIGN_OUT_BOTTOM_MID, 0,0);

    keypad_update(group);/* What is doing this function? See below... */
}
    
    
void keypad_update(lv_group_t * keypad_group)
{
    lv_indev_set_group(kp_indev, keypad_group);
}
```

Could you please try this code and report if you are able to reproduce the problem?

kisvegabor, 2018-11-09:
Hi,

I tried your code in the PC simulator and I didn't see any issues. See the video: [group_issue.mp4.zip](https://github.com/littlevgl/lvgl/files/2565658/group_issue.mp4.zip)

I used the `keyboard` driver like this:
```c
    lv_indev_drv_t kb_drv;
    kb_drv.type = LV_INDEV_TYPE_KEYPAD;
    kb_drv.read = keyboard_read;
    kp_indev = lv_indev_drv_register(&kb_drv);
```

TMarkivULCRobotics, 2018-11-09:
Can you resend the video bacause it is 0.021 s long ?

TMarkivULCRobotics, 2018-11-09:
Did you use the `lv_indev_set_group(kp_indev, keypad_group);` function call ?

TMarkivULCRobotics, 2018-11-09:
My understanding is that the 'group' is the mechanism for  the keypad. When I register the group in the keypad indev structure (using `lv_indev_set_group` function) then the library allows to move the 'focus', between objects in the particular group, using keypad. Since lvgl v5.3 I see that the touchpad press activates (makes the object focused) the object if it is registered in the  group.

If, for example, I have a message box, then I create a new group which contains buttons from the message box. After the group is created then I have to put that new group into keypad indev structure using the  `lv_indev_set_group` function call. So that I can navigate between message box objects.

So actually I have some amount of keypad groups with different objects added to them. And I am registering those group in the keypad indev structure in different places of the application in run-time.

Also what I found is that the only way to edit text area is to add the text area object into some group and register that group in input device structure.

Could you please confirm if my understanding is correct?

kisvegabor, 2018-11-10:
Video:
https://drive.google.com/open?id=1w4qvDmF8Uiu684q9y32nj4tdDj89GGTw

Focus with click:
Yes, if the object is in a group, a click will make it focused. You can disable it with `lv_group_set_click_focus(group, false)`
 
> If, for example, I have a message box, then I create a new group which contains buttons from the message box. After the group is created then I have to put that new group into keypad indev structure using the lv_indev_set_group function call. So that I can navigate between message box objects.

Almost. If you created an `lv_mbox` then you need to add the mbox object to the group and not its buttons directly. The mbox will handle it. However, if you created a container, labels and buttons yourself then you should the buttons individually.  Have look at the keyboard tutorial: https://github.com/littlevgl/lv_examples/blob/master/lv_tutorial/10_keyboard/lv_tutorial_keyboard.c

Maybe I misunderstood something but why do you want to create a new group for every object? If you have a keypad then one group should be enough for that.

TMarkivULCRobotics, 2018-11-10:
I have few screens and a tabview. Each of them has a group that include all objects on the particular  tab or screen.

kisvegabor, 2018-11-10:
It shouldn't cause the issue, but I don't understand something:
In the video you sent I see that you use two groups: one for the text area and an other for the tabview. What is the purpose of using more groups if you have only one input device?

TMarkivULCRobotics, 2018-11-12:
I have one group for each tab in the tabview object. This is to prevent navigating through objects which are not visible on the particular tab or screen. In the tabload function I always update the keypad with the particular group using `lv_indev_set_group(kp_indev, keypad_group);`

Regarding mbox - no question about it. I am doing just like you described.

Could you please review the following example?
I have a `tabview` with `tab1` and `tab2`. `tab1` has `text1` and `btn1` objects. `tab2` has `text2` and `btn2` objects. How to make proper keypad navigation through all this objects?

My solution is the following pseudocode :
```
/* Screen Initialization function.  */
{
    group1 = lv_group_create();
    lv_group_add_obj(group1, tabview);
    lv_group_add_obj(group1, text1);
    lv_group_add_obj(group1, btn1);

    group2 = lv_group_create();
    lv_group_add_obj(group2, tabview);
    lv_group_add_obj(group2, text2);
    lv_group_add_obj(group2, btn2);
}

/* Tabview load callback function. */
lv_res_t job_tab_loaded(lv_obj_t *tabview, uint16_t tab_id)
{
    switch (tab_id)
    {
    case 0 : keypad_update(group1);break;
    case 1 : keypad_update(group2);break;
    }
}
```

What do you think about this?

kisvegabor, 2018-11-14:
Your code looks good. The only thing I don't understand is that in the video there was two focused objects (orange colored):
1. The tab itself
2. And the text area

In your described method there should be only one focused object.

[image]

TMarkivULCRobotics, 2018-11-17:
Hello Kiss, I get how to avoid the problem. 

In the keypad read function I added the code that clears the `lv_indev_data_t * data` input parameter structure.
Also I prepared the project that I think you may easily use to check the issue. This is the project that is from the lv_boards.zip file that you made. The target device is STM32F746 disco.

In the attached archive there is a project, and in the main.c file you'll find the following code:
```
bool keypad_read(lv_indev_data_t * data)
{
#if REPRODUCE_THE_ISSUE == 1
    /* Do not clear the *data structure */
#else
    data->state = LV_INDEV_STATE_REL;
    data->key = 0;
#endif
    return false;
}
```

In the previous implementation I did not touch the input parameter if the keypad controller did not output any data. But now I always clear the input parameter.

Is that behavior expected? 
[stm32f746_discovery_no_os.zip](https://github.com/littlevgl/lvgl/files/2591279/stm32f746_discovery_no_os.zip)

kisvegabor, 2018-11-18:
Yes, it really could cause the issue. In the read function, you always should set the related fields of `data`. 
However the examples you can find set the fields it should be:
1. More emphasized in the documentation
2. The library should do something to avoid this behavior. See: #560

kisvegabor, 2018-11-22:
I merged the fix into master and dev-5.3.

If you are still experiencing the issue, please reopen this issue

closed 2018-11-22 by kisvegabor