lvgl/lvgl #473: Bigger font without memory impact (scale at runtime?)

AloyseTech, 2018-10-22:
Hi,

I need to display huge numbers on a small mcu project. I created a 80p font and it works on the pc simulator, but storing this font takes way too much flash (obviously). Is there a way to scale the text at runtime? (ie use a smaller font but render it x times bigger)

AloyseTech, 2018-10-23:
I think this might be related to #107 and vector fonts. Do you have a code example?

kisvegabor, 2018-10-27:
Hi,

There is no font scale feature right now. :(

I have some tips to reduce the memory usage:
-  use smaller bpp (bit-per-pixel)
- use the range what you really need
- use the list of characters what you really need

The vector fonts can also work but there are no examples yet. However, it would be great to have examples :) If you want to try it I kindly help you if you stuck somewhere.

AloyseTech, 2018-11-12:
I quickly looked into this but I think this is a ad idea to use ttf font rendering on the fly, given the size of a ttf file...

embeddedt, 2018-11-12:
@AloyseTech Agreed. The repeated computational overhead of generating the bitmaps is also unsuitable for an embedded system, since there is little to no RAM available for caching.

kisvegabor, 2018-11-14:
I can imagine that you store the TTF font in an external flash or SD card. If the TTF file has changed you can generate the required fonts in the flash of the MCU.

AloyseTech, 2018-11-14:
> 
> 
> I can imagine that you store the TTF font in an external flash or SD card. If the TTF file has changed you can generate the required fonts in the flash of the MCU.

Storing the rendered font bitmaps on the SD card is better then I guess.

kisvegabor, 2018-11-15:
> Storing the rendered font bitmaps on the SD card is better then I guess.

I think it might be slow to read the SD for every letter.

AloyseTech, 2018-11-15:
IMO the whole point of run-time rendering is to avoid consuming flash with all the pre-rendered fonts. If external memory is available it might not be needed, given the amount of storage on an SD card (or any external flash nowadays). It could be useful if the TTF file is smaller than all the rendered fonts using this TTF and the rendering could be directly pushed to the VDB or frame buffer (if fast enough, I have no idea how complex it is to render TTF font).

I've not yet tried it (still hesitating to buy a STM32H7 or F7 board...), but reading bitmaps from an SD card on an SDIO 4bit bus at around 100MHz should be quite efficient I guess, don't you think?

kisvegabor, 2018-12-18:
I close this issue because there was no activity here for a while.

If you have any news on this topic feel free to comment here and reopen the issue.

closed 2018-12-18 by kisvegabor