lvgl/lvgl #474: Group Focus Policy on Deletion

BrianPugh, 2018-10-23:
I was wondering if we could add an option to lv_conf.h on how the next object should be focused on deletion?

Currently the focus is set on line 125 of [lv_group.c](https://github.com/littlevgl/lvgl/blob/master/lv_core/lv_group.c):
```
    if(*g->obj_focus == obj) {
        lv_group_focus_next(g);
    }
```
In my case, I'd much prefer to focus on the previous object in the group on deletion, perhaps something like this:
```
    if(*g->obj_focus == obj) {
#if LV_REFOCUS_POLICY == 0
        lv_group_focus_next(g);
#elif LV_REFOCUS_POLICY == 1
        lv_group_focus_prev(g);
#endif
    }
```

Full scenario: I have a menuing system on a small screen where each submenu takes up the entire screen. Whenever I press my back button, I delete the currently focused submenu. If I am 3 submenus deep, deleting the currently focused submenu will cause the 1st submenu to be in focus while the 2nd submenu is the one displayed. Changing the focus policy to above causes my desired behavior.

ederguidsa, 2018-10-25:
In my opinion, such option should be group specific, not a "#define" on lv_conf.h. You may want different behaviours for different groups.

BrianPugh, 2018-10-25:
How about adding a new attribute  `uint8_t refocus_policy :1` to the `lv_group_t` struct thats initially set to 0; probably make an enum like:

```
typedef enum lv_group_refocus_policy_t {
    LV_GROUP_REFOCUS_POLICY_NEXT = 0,
    LV_GROUP_REFOCUS_POLICY_PREV = 1
} lv_group_refocus_policy_t;
```

We can then add a function like
```
void lv_group_set_refocus_policy(lv_group_t *g, lv_group_refocus_policy_t policy) {
    g->refocus_policy = policy;
}
```
And finally
```
    if(*g->obj_focus == obj) {
        if(g->refocus_policy == LV_GROUP_REFOCUS_POLICY_NEXT)
            lv_group_focus_next(g);
        else if(g->refocus_policy == LV_GROUP_REFOCUS_POLICY_PREV)
            lv_group_focus_prev(g);
        else
            // raise some error
            ;
    }
```

BrianPugh, 2018-10-29:
Feature implemented in #495

closed 2018-10-29 by BrianPugh