lvgl/lvgl #476: Process for generating lv_font_symbol of custom size

AloyseTech, 2018-10-24:
Hello,

Could you please describe the way you generate the symbol font using the online converter please? I'm unable to make it work with a custom size. The standard font works in 16px.

Here is what I've done : 
1. Use FontAwesome.ttf
2. Range from 61440 to 62190
3. 16px
4. BPP to 1 (will be built-in font)
5. "Monospace", "Scale" and "Base Line offset" left empty
6. List copied from the useful note at the bottom of the page
7. Check "Convert as built-in font"

I successfully got the c file, named lv_font_symbol_16.c, I copy it in my project root folder, next to main. I change the include from "../lv_misc/lv_fonts.h" to "lvgl/lv_misc/lv_fonts.h".

In my main.c file, I add the following code after lv_init()
```
#if USE_LV_FONT_SYMBOL_16 != 0
#if USE_LV_FONT_DEJAVU_16 != 0
	lv_font_add(&lv_font_symbol_16, &lv_font_dejavu_16);
#else
	lv_font_add(&lv_font_symbol_16, NULL);
#endif
#endif
```
And in lv_conf.h I have : 
```
#define USE_LV_FONT_DEJAVU_16		4
#define USE_LV_FONT_SYMBOL_16	4
#define LV_FONT_CUSTOM_DECLARE	LV_FONT_DECLARE(lv_font_dejavu_16)\
								LV_FONT_DECLARE(lv_font_symbol_16)
```

Text is working fine but the symbols are not displayed in a label.... I tested with 10px and 20px and it works, so usage should be fine.

I tried to edit the font source to match the built-in symbol source file, i.e. changing from sparse mode to continuous (rename the get_bitmap and get_width, and set unicode_list to NULL), but with no luck also...

AloyseTech, 2018-10-25:
I made it work by manually replacing the unicode_first and unicode last with LV_SYMBOL_GLYPH_FIRST and LV_SYMBOL_GLYPH_LAST (and keeping the change for continuous mode instead of sparse).
Maybe a checkbox could be added to the online converter for symbols?

AloyseTech, 2018-10-28:
This is a good idea yes. So you can change the way you generate your symbols without the user noticing it eventually.

kisvegabor, 2018-10-28:
I've just added the checkbox at the bottom of the form. Can you verify it really works?

kisvegabor, 2018-10-29:
Upps, I forget to upload a file. Should work now.

AloyseTech, 2018-10-29:
Yep it seems to work now!

kisvegabor, 2018-10-29:
Great! :)
Thank you for the suggestion!

closed 2018-10-29 by kisvegabor