lvgl/lvgl #477: input driver callback doesn't execute by lv_task_handler

ImmortalBuka, 2018-10-24:
stm32f4
```
init:
lv_indev_drv_t indev_drv;
lv_indev_drv_init(&indev_drv);
indev_drv.type = LV_INDEV_TYPE_POINTER;
indev_drv.read = my_input_read;
lv_indev_drv_register(&indev_drv);
```
and in debugger i don't catch executing my_input_read().

ederguidsa, 2018-10-24:
After registering the driver, you must also call "lv_indev_init();"
Also, before the "init:" label you must initialize your hardware.

**Example (using linux 'evdev' driver)**

evdev_init(); //Initialize linux interface with 'evdev' (in your case, you would initialize your hardware)
lv_indev_drv_t  indev_drv;
lv_indev_drv_init(&indev_drv);
indev_drv.type = LV_INDEV_TYPE_KEYPAD;
indev_drv.read = evdev_read;
indev = lv_indev_drv_register(&indev_drv);
lv_indev_init(); //initialize lvgl input task

ImmortalBuka, 2018-10-24:
I'll try. Where in docs i can find information about it?

embeddedt, 2018-10-24:
Do you call `lv_tick_inc`?

ImmortalBuka, 2018-10-25:
> 
> 
> Do you call `lv_tick_inc`?

thx, it works!
but, why it not present in examples?

embeddedt, 2018-10-25:
@ImmortalBuka Which example is missing it? It states on the [Porting documentation page](
https://github.com/littlevgl/lvgl/wiki/Porting#Tick-interface) that you need to use that function.