lvgl/lvgl #480: Creating gray background for message boxes doesn't work in lvgl 5.2 and higher

TMarkivULCRobotics, 2018-10-24:
Hello,

The following code is used for displaying message box with gray background:
```
void ui_display_warning_message(const char * msg)
{
    static lv_style_t gray_style;
    lv_style_copy(&gray_style, &lv_style_plain);
    gray_style.body.opa = LV_OPA_70;
    gray_style.body.main_color = LV_COLOR_HEX3(0x111);
    gray_style.body.grad_color = LV_COLOR_HEX3(0x111);

    lv_obj_t * gray_bg = lv_obj_create(lv_scr_act(), NULL);
    lv_obj_set_style(gray_bg, &gray_style);
    lv_obj_set_size(gray_bg, LV_HOR_RES, LV_VER_RES);

    static const char * btns[] = {"Ok", ""};

    lv_obj_t * mbox = lv_mbox_create(gray_bg, NULL);
    lv_mbox_set_text(mbox,msg);
    lv_mbox_add_btns(mbox, btns, common_mbox_callback);
    lv_obj_align(mbox, NULL, LV_ALIGN_CENTER, 0, 0);

        prev_group = keypad_get_group();
	mbox_key_group = lv_group_create();
	lv_group_add_obj(mbox_key_group, mbox);
	keypad_update(mbox_key_group);
}
```
The background was successfully generated in LVGL5.1.1. Since 5.2 update that code doesn't make gray background.

How to make gray background in LVGL5.3?
Where are release notes for each library version?

Thanks,
Taras

kisvegabor, 2018-10-28:
Hi Taras,

I simplified your code to test it:
```c
void ui_display_warning_message(const char * msg)
{
    static lv_style_t gray_style;
    lv_style_copy(&gray_style, &lv_style_plain);
    gray_style.body.opa = LV_OPA_70;
    gray_style.body.main_color = LV_COLOR_HEX3(0x111);
    gray_style.body.grad_color = LV_COLOR_HEX3(0x111);

    lv_obj_t * gray_bg = lv_obj_create(lv_scr_act(), NULL);
    lv_obj_set_style(gray_bg, &gray_style);
    lv_obj_set_size(gray_bg, LV_HOR_RES, LV_VER_RES);

    static const char * btns[] = {"Ok", ""};

    lv_obj_t * mbox = lv_mbox_create(gray_bg, NULL);
    lv_mbox_set_text(mbox,msg);
    lv_mbox_add_btns(mbox, btns, NULL);
    lv_obj_align(mbox, NULL, LV_ALIGN_CENTER, 0, 0);
}
```
And it worked for me using the latest `master` branch.

v5.3 was a mistake. The current version is "v5.2 release candidate" available in the master branch.

The release notes are here: https://github.com/littlevgl/lvgl/releases

TMarkivULCRobotics, 2018-10-29:
Do you mean that there is a mistake in the v5.3 ?

kisvegabor, 2018-10-30:
Sorry, I wasn't clear. There were a few days while the current version in master (lvgl.h) was v5.3 but it was just a mistake during merging branches. So:
1. The current version is `v5.2-rc` in `master`
2. The current development branch is `dev-5.3`

TMarkivULCRobotics, 2018-11-26:
This was a bug in my implementation of the GPU functions. No problem with the library.

kisvegabor, 2018-11-27:
Glad to hear that you solved it!