lvgl/lvgl #481: Roller object doesn't work in LVGL5.3 dev

TMarkivULCRobotics, 2018-10-24:
The roller object doesn't keep the value that is selected using touchpad or keypad.

kisvegabor, 2018-10-28:
I tested it and it worked for me. Are you using the latest version from the `dev-5.3 ` branch?

TMarkivULCRobotics, 2018-10-29:
It seems that I must press LV_GROUP_KEY_ENTER to save the value in the roller.

But I remember that in the 5.2 version, when I been using the touch pad then I had not a need to press LV_GROUP_KEY_ENTER. But there were another issue - it was not possible to setup the roller value using keypad.

Now it looks like the problem was moved from keypad to touchpad)

TMarkivULCRobotics, 2018-10-29:
Yes I just updated my LVGL 5.3  version.

kisvegabor, 2018-10-30:
> It seems that I must press LV_GROUP_KEY_ENTER to save the value in the roller.

Yes, this is the intended behavior. It gives the opportunity to stay with the current value on defocus.
 
> But I remember that in the 5.2 version, when I been using the touch pad then I had not a need to press LV_GROUP_KEY_ENTER. But there were another issue - it was not possible to setup the roller value using keypad.

I pushed a fix in `master` and `dev-5.3`. Now you can select a new value with touchpad without pressing ENTER and/or keypad by pressing ENTER.

embeddedt, 2018-11-23:
There was no activity here for 1 month so this issue has been closed. If you continue to have problems please reopen the issue.

closed 2018-11-23 by embeddedt