lvgl/lvgl #482: lv_ddlist: LV_LABEL_LONG_ROLL and LV_LABEL_LONG_DOT effects not as expected

ederguidsa, 2018-10-25:
I have the following scenario: 

I'm using a drop down list to list some options. These options may have a very long number of characters. So, I was expecting to use the LV_LABEL_LONG_ROLL or LV_LABEL_LONG_DOT in order to print the long options beautifully.

When I try to apply these effects to the drop down list,  all the options are considered as an unique label, and I don't get the desired result. I've checked the code and concluded that indeed all options are stored in an unique label. 

Is there any workaround to apply these effects considering each option as a separate label?

embeddedt, 2018-10-28:
To build on what @kisvegabor said, you might be able to attach a custom signal function, and use that to "open" the list when it is clicked (adjust its height). Completely untested, just a thought.

embeddedt, 2018-11-22:
There was no activity here for 1 month so this issue has been closed.

closed 2018-11-22 by embeddedt