lvgl/lvgl #483: Tabview load action function prototype  has changed in new LVGL version

TMarkivULCRobotics, 2018-10-25:
In the 5.1.1 version tabview load action function prototype was the following:
`typedef lv_res_t (*lv_tabview_action_t)(lv_obj_t *, uint16_t);`
Returning LV_RES_INV, in the tab load action function, was convenient to prevent particular tab from loading.

In the 5.3 dev version the prototype is the following:
`typedef void (*lv_tabview_action_t)(lv_obj_t *, uint16_t);`
Function doesn't return any value and tab load prevention method from 5.1.1 doesn't work.

The question is how to prevent particular tab from loading when it is needed in LVGL5.3 ?

kisvegabor, 2018-10-28:
I didn't consider this kind of use of the return value.
Anyway, it sounds reasonable so I reverted it: https://github.com/littlevgl/lvgl/commit/fc319a45024f7fa7ce1a86385e731142590ad2d3

TMarkivULCRobotics, 2018-10-29:
Thank you very much for that!)

closed 2018-10-29 by TMarkivULCRobotics

kisvegabor, 2018-10-29:
Your welcome! :)