lvgl/lvgl #484: Style of box around selected object in group

AloyseTech, 2018-10-26:
I seems that there is no way to change the style of the "box" around a selected object in a group. It is possible to change the style of the object when selected though, which is nice and could be used for creating nice effects. Would it be possible to add a lv_group_set_selection_style or something similar to change the style of the selected box (or even hide it) ?

ederguidsa, 2018-10-26:
Do you mean the focused object in a group? 
If so, see https://littlevgl.com/basics, in the section 'Touchad-less navigation'. There is a callback function called '**lv_group_set_style_mod_cb**' that allows you to change the style of the box that indicates the focused object.

embeddedt, 2018-10-26:
We have been moving the LVGL documentation to GitHub. That page will be taken offline at some point I presume, in which case you can find it here:

https://github.com/littlevgl/lvgl/wiki/Input-devices#Touchpad-less-navigation

AloyseTech, 2018-10-26:
I'm trying to only change the color of the text inside a Text Area which has nothing else but the text displayed. When the object is selected, I have a back border and a white body color, even if I use the following set_style_mod : 

```C
static void styleMod(lv_style_t * style)
{
	style->text.color = LV_COLOR_MAKE(0x70, 0xFF, 0x70);
	style->body.empty = true;
	style->body.border.width = 0;
}
```

AloyseTech, 2018-10-26:
> 
> 
> Do you mean the focused object in a group?
> If so, see https://littlevgl.com/basics, in the section 'Touchad-less navigation'. There is a callback function called '**lv_group_set_style_mod_cb**' that allows you to change the style of the box that indicates the focused object.

I think the lv_group_set_style_mod_cb let you change the style of the selected obect, not the box that is added around. I might be wrong though, I'm new to lvgl.

AloyseTech, 2018-10-28:
Here is the detailed code : 
```C
	static lv_style_t screenStyle;
	lv_style_copy(&screenStyle, &lv_style_plain);
	screenStyle.body.radius = 0;
	screenStyle.body.padding.hor = 0;
	screenStyle.body.padding.ver = 0;
	screenStyle.body.padding.inner = 0;
	screenStyle.body.border.width = 0;
	screenStyle.body.shadow.width = 0;
	screenStyle.body.main_color = LV_COLOR_MAKE(0x56,0x56,0x56);
	screenStyle.body.grad_color = LV_COLOR_MAKE(0x44,0x44,0x44);

	static lv_style_t labelStyle;
	lv_style_copy(&labelStyle, &lv_style_plain);
	labelStyle.text.font = &code_new_roman_digit_bold_80;//&dejavu_bold_80;
	labelStyle.text.color = LV_COLOR_MAKE(0xF0, 0xF0, 0xF0);
	labelStyle.body.border.width = 0;
	labelStyle.body.padding.ver = 0;
	labelStyle.body.padding.hor = 0;
	labelStyle.body.padding.inner = 0;
	labelStyle.body.empty  = true;

	lv_obj_t * screen1 = lv_cont_create(NULL, NULL);
	lv_obj_set_style(screen1, &screenStyle);
	lv_obj_set_size(screen1, 320, 240);

	lv_obj_t * labelMeasV = lv_label_create(screen1, NULL);
	lv_ta_set_text(labelMeasV, "+03.298");
	lv_ta_set_style(labelMeasV, LV_TA_STYLE_BG, &labelStyle);

	lv_obj_t * group1 = lv_group_create();
	lv_indev_set_group(encoder_indev, group1); //set the encoder as the group input device
	lv_group_set_style_mod_cb(group1, styleMod);
```

```C
static void styleMod(lv_style_t * style)
{
	style->text.color = LV_COLOR_MAKE(0x70, 0xFF, 0x70);
	style->body.empty = true;
	style->body.border.width = 0;
}

![stylemodissue1](https://user-images.githubusercontent.com/13847288/47616312-67cc9380-dabb-11e8-92c2-323cc5fb6776.PNG)

```

AloyseTech, 2018-10-29:
Thanks it works as expected now. :)

closed 2018-10-29 by AloyseTech