lvgl/lvgl #485: New object idea : number spin box

AloyseTech, 2018-10-26:
I'm looking for a way to display and enter number in a given range. I'm used to Qt and the QSpinBox object. It seems that it could be an extension of a text area, but with a validator for numeric characters only and special action on some keys (KEY_UP and KEY_DOWN would increase/decrease the number by a given STEP amount, which could be an integer of a float).

embeddedt, 2018-10-26:
You can use `lv_ta_set_accepted_chars` to set the list of accepted characters to digits only.

There isn't an API for changing the behavior of the up and down keys, but you could write your own signal function that checks for the up or down keys before passing the result to the original ancestor signal function.

Have a look at the text area signal function here:

https://github.com/littlevgl/lvgl/blob/88b8e982b485368cc873860ba602a4cd315d674f/lv_objx/lv_ta.c#L1084

And lastly, if you wanted a more classic spinbox, you could add two buttons and give them up and down arrows.

AloyseTech, 2018-10-26:
I already have a validator for the charset (""1234567890.+-"). I will investigate the signal handler modification, thanks

kisvegabor, 2018-10-28:
I agree with @embeddedt . the signal function can be used to overwrite the current behavior. 

@AloyseTech If you are interested you can even add a new object type based on the Text area. However, modifing the signal function is much more simple.

AloyseTech, 2018-10-28:
I'm actually trying to create a new object, since it seems to be more appropriate since the spinbox should have specific additional parameters.

embeddedt, 2018-10-28:
@AloyseTech Are you making a new object with a custom design function, or making a new object that will have a button, text area, etc. as children?

AloyseTech, 2018-10-28:
I’ve made a copy of the ta object and started modifying it. I’ve not been far at the moment though. Only modified the structure parameter and getter/setters.

kisvegabor, 2018-10-28:
@embeddedt I agree. If a new object type should work like an existing one than the new object should be inherited from the existing.  In this case, the Text area seems a good ancestor only it's signal function need to be modified.

AloyseTech, 2018-10-31:
I have created an object which extends Text Area (as ancestor). Here is what I've got at the moment :
[image]
I use an encoder as input device.

kisvegabor, 2018-10-31:
@AloyseTech Wow, looks awesome! Can you create a PR?

AloyseTech, 2018-11-01:
@kisvegabor thanks. I will create a PR

AloyseTech, 2019-02-01:
It is in the documentation. The doc page is not complete but the test code is available.