lvgl/lvgl #487: Is there existed API to blink a text?

TheShark90, 2018-10-26:
Hi @kisvegabor ,

As I'm a student in this library, with trying to blink a text but seems there is not any existed API to do that.
Is this correct?

Please teach me. Thank you for your support.
Regards.

embeddedt, 2018-10-26:
One way to blink text could be to use an LVGL animation, and use `lv_obj_set_hidden` as your function to animate the object. The `start` value would be `false` and the `end` value should be `true`.

This is untested and just a concept, but it might help.

TheShark90, 2018-10-26:
Thank you @embeddedt ,

I think it's extremely clear and the best way, maybe I will set a timer between start-end

kisvegabor, 2018-10-28:
FYI: https://github.com/littlevgl/lvgl/wiki/Animations

robertvxt, 2018-10-31:
Thanks,
I followed to @embeddedt 's suggestion then the text can be blinked.

But would you give me advice in case: just blink partially of text?
i.e. The test "Yes/No", just blink "Yes" and fix "/No" part

My proposal, break the text to 2 separate objects: "Yes" and "/No". Set their position (same y coord.) then just blink.

Let me re-open the ticket

embeddedt, 2018-10-31:
Yes, using two objects is probably the simplest method. Or you could try recoloring the text to be the background color (see https://github.com/littlevgl/lvgl/wiki/Label).

kisvegabor, 2018-11-02:
> My proposal, break the text to 2 separate objects: "Yes" and "/No". Set their position (same y coord.) then just blink.

It would be a clear solution. 
The solution of @embeddedt also works if the background is homogenous. In addition, it consumes less RAM.

robertvxt, 2018-11-18:
Thanks @embeddedt , It's quite good solution when we use fixed options (YES/NO)
For my case, I want to make it becomes common for other texts... so I will choose to break text into separate objects.

closed 2018-11-18 by robertvxt