lvgl/lvgl #488: Scroll while touching

bernpuc, 2018-10-26:
Hi All,

I have a list of labels in a page which I want to be able to scroll. Currently, I can only get the list to scroll if I touch the screen, drag my finger, then release. The scrolling only happens once I release. What I would like is to be able to touch, drag to scroll, then release. Is there something I have to configure in the list object? Or is it something to do with my touchpad read function?
```C
  /* Scrollable List of script names */
  lv_obj_t* script_list;
  script_list = lv_list_create(scriptSelect.scr, NULL);
  lv_obj_set_width(script_list, lv_obj_get_width(lv_obj_get_parent(script_list)));
  lv_obj_set_height(script_list, lv_obj_get_height(lv_obj_get_parent(script_list)) - 55);
  lv_obj_align(script_list, NULL, LV_ALIGN_IN_TOP_MID, 0, 0);
```

```C
bool ex_tp_read(lv_indev_data_t *data)
{
  bool retval;
  
  /* Read your touchpad */
  touchUpdateState();
  touchGetCalPos(&touch_data);
  /* data->state = LV_INDEV_STATE_REL or LV_INDEV_STATE_PR */
  /* data->point.x = tp_x; */
  /* data->point.y = tp_y; */
  data->point.x = touch_data.cal_x;
  data->point.y = touch_data.cal_y;
  
  if (touch_data.touch_state.touch_detected)
  {
    data->state = LV_INDEV_STATE_PR;
    retval = TRUE;
    cmdResetIdleTimer();    /* Any touch when idle should reset idle timer */
  }
  else
  {
    data->state = LV_INDEV_STATE_REL;
    retval = FALSE;   /*false: no more data to read because we are no buffering*/
  }
  return retval;
}
```

embeddedt, 2018-10-26:
Issue #204 might be helpful.

kisvegabor, 2018-10-28:
I see some issued in your TP read function:
1. In releases state `data->point.x/y` should be set to the last pressed value.
2. You should return TRUE only if there is more data to read (you have buffered some data). In your case, it seems you should return FALSE in every case.

fallstool, 2018-10-29:
I think PressAction and relaease action only shoud once.
try this code:

`bool retval = FALSE;
bool ex_tp_read(lv_indev_data_t *data)
{

  /* Read your touchpad */
  touchUpdateState();
  touchGetCalPos(&touch_data);

 
  if (touch_data.touch_state.touch_detected)
  {
    data->point.x = touch_data.cal_x;
	data->point.y = touch_data.cal_y;
    if(!retval)
    {
        data->state = LV_INDEV_STATE_PR;
    	retval = TRUE;
    }    
  
    cmdResetIdleTimer();    /* Any touch when idle should reset idle timer */
  }
  else
  {
    data->state = LV_INDEV_STATE_REL;
    retval = FALSE;   /*false: no more data to read because we are no buffering*/
  }
  return retval;
}`

embeddedt, 2018-11-22:
There was no activity here for 1 month so this issue has been closed. If you continue to have problems please reopen the issue.

closed 2018-11-22 by embeddedt

bernpuc, 2019-01-03:
Sorry for not responding sooner. Thank you for your suggestions which have addressed my scrolling issue.

embeddedt, 2019-01-03:
@bernpuc Glad to hear it!