lvgl/lvgl #489: suggestion: halftone object fill?

ripe909, 2018-10-28:
Using a monochrome display, a halftone fill would be very useful...  maybe part of the style instead of a gradient using the body.main_color and body.grad_color for the ?

example:
[image]

kisvegabor, 2018-10-28:
It sounds great! A Pull request would be welcome.

I think it can be added to the `sw_color_fill` function in `lv_draw_vbasic`.

As this feature can work only for monochrome displays it can be activated if the color is `LV_COLOR_GRAY` and `LV_COLOR_DEPTH == 1`.  What do you think?

The implementation seems simple too. E.g. all odd x and odd y coordinates should be black, the others are white.

So are you interested in the implementation?

ripe909, 2018-10-29:
I tried something similar, but I think the "even" and "odd" changes relative to the object being drawn, so it ends up making a mess (for example as the mouse pointer moves over the screen and redraws.

example:
[image]

kisvegabor, 2018-10-29:
Awesome!

`fill_area` really contains relative coordinates. To make them absolute use `mem_area.x1 + fill_area.x1` and `mem_area.y1 + fill_area.y1`

kisvegabor, 2018-11-21:
Hi,
Is there any news here?

kisvegabor, 2018-11-29:
I close this issue because there was no activity here for a while .

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2018-11-29 by kisvegabor