lvgl/lvgl #490: suggestion: dashed/dotted lines

ripe909, 2018-10-28:
Similar to the halftone request, creating dotted/dashed lines using style would be very useful, maybe specify the two colors and the dash size:

example:
[image]

kisvegabor, 2018-10-28:
Hi,

Good idea as well!

It seems simple to implement for the horizontal and vertical lines but quite complicated for the skewed lines. Anyway, dashed/dotted support for the straight lines is better than no support at all.

You can see the relevant code in `lv_draw_line.c` in `line_draw_hor` and `line_draw_ver` function.

Would you like to add this feature?

kisvegabor, 2018-11-30:
I close this issue because there was no activity here for a while .

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2018-11-30 by kisvegabor